Polidea/RxBluetoothKit #132: Cannot subscribe to Observable<Characteristic> - timeout error.

mxndev, 2017-08-14:
I have a problem with subscribe multiple different characteristic types in my application. At first I connect to the device and declare Observable values
`let peripheralObservable <- let with information about connected device
let dataStream1Observable = peripheralObservable
        .flatMap { (peripheral) -> Observable<Characteristic> in
            return peripheral.setNotificationAndMonitorUpdates(for: "some ID")
        }.map { (characteristic) -> Data? in
            return characteristic.value
        }
        
    let isConnectedObservable = peripheralObservable
        .flatMap { (peripheral: Peripheral) -> Observable<Bool> in
            return peripheral.rx_isConnected
        }`

Then I subscribed first value (isConnectedObservable), everything works good. Next I call dispose on this subscribtion and start new one on dataStream1Observable, then I wait a moment and I get error "timeout error". What I do wrong? It is a error in my application or in this library. I really need help.

Kacper20, 2017-08-23:
Hello @mxndev,

I'm looking into it, thanks for the patience

Kacper20, 2017-08-23:
Hi @mxndev,

Could you please show how `peripheralObservable` looks like and reveal exact error? On which observable are you doing subscription?

Kacper

mxndev, 2017-08-23:
Value of `peripheralObservable` is returned by function posted below:
`private let manager = BluetoothManager(queue: .main, options: nil)

    func observablePeripheral(serviceUUIDs: [CBUUID]?) -> Observable<Peripheral> {

    let peripheralObservable = manager.rx_state
        .filter { $0 == .poweredOn }
        .timeout(1.0, scheduler: MainScheduler.instance)
        .take(1)
        .flatMap { _ in
            self.manager.scanForPeripherals(withServices: serviceUUIDs)
        }
        .take(5.0, scheduler: MainScheduler.instance)
        .toArray()
        .flatMap({ (scannedPeripherals) -> Observable<[ScannedPeripheral]> in
            return scannedPeripherals.isEmpty ? .error(RxError.timeout) : .just(scannedPeripherals)
        })
        .map {
            return $0.sorted(by: { (lhs, rhs) -> Bool in
                return lhs.rssi.doubleValue > rhs.rssi.doubleValue
        })
        }
        .flatMap {
            Observable.from($0)
        }
        .take(1)
        .flatMap( { (s:ScannedPeripheral) -> Observable<Peripheral> in
            return s.peripheral.connect()
        })

        return peripheralObservable
    }`

The error is:
`Received unhandled error: ~/SignalStreamProvider.swift:134:subscribeToDataStreamObservables() -> Sequence timeout.`
I don't understand second question, as I said first I make a subscription on isConnectedObservable and then after dispose of that subscription I make subscription on dataStream1Observable.

Kacper20, 2017-08-23:
@mxndev Okay, I'm starting to understand. Could you please post exact type of an error? Is it RxSwift error? You should be able to print it or log in lldb

mxndev, 2017-08-23:
I post it earlier:`Received unhandled error: ~/SignalStreamProvider.swift:134:subscribeToDataStreamObservables() -> Sequence timeout`
I don't know, how can I get more information about this error.

Kacper20, 2017-08-23:
How are you handling the error? You should have it available in the `subscribe(onError:` block.)

mxndev, 2017-08-23:
Getting onError callback I get only `Sequence timeout.` on logs.

Kacper20, 2017-08-23:
And what if you inspect the error instance received in `onError` call in debugger? What type is it?

mxndev, 2017-08-23:
I see in debuger 'timeout' type of error. `error	RxSwift.RxError	timeout`

Kacper20, 2017-08-23:
Great! :)

It's the result of `timeout` operator being used. Cause is probably here:

```
let peripheralObservable = manager.rx_state
    .filter { $0 == .poweredOn }
    .timeout(1.0, scheduler: MainScheduler.instance)
```
Could you add call to RxSwift debug after that and after `manager.rx_state` and send me the logs? :)

mxndev, 2017-08-23:
I'm really new in RxSwift and I don't know how to subscribe to error. I check some thing, when I commit this line with .timeout option, I still get timeout error. Definitely this isn't the reason.

Kacper20, 2017-08-23:
Are you having timeout calls anywhere else in your source code?

Kacper20, 2017-08-23:
Also, in your code you're reporting `RxSwift.timeout` error explicitely, here:
```
    .flatMap({ (scannedPeripherals) -> Observable<[ScannedPeripheral]> in
        return scannedPeripherals.isEmpty ? .error(RxError.timeout) : .just(scannedPeripherals)
    })
```
I believe you should return .empty() instead of an error, because scanning fires multiple times.

mxndev, 2017-08-23:
Ok, when I removed it, no error is displayed, but still my subscription doesn't work and I don't get any data from bluetooth device.

Kacper20, 2017-08-24:
@mxndev I can't do any more until you provide me some additional info(when it happens, what are the logs)

mxndev, 2017-08-24:
@Kacper20 sorry, but I don't know, what can I say more. I said earlier, that happen, when I subscribe second observer after some time elapsed from first one. In logs I didn't get any data.

mxndev, 2017-08-31:
Great, you leave this error without any reply? Why are you don't want to try repeat that error in your codebase? It's really frustrating.

Kacper20, 2017-09-06:
Hi @mxndev,

It's not error in our library. It's probably bug in your code. I've asked for some logs and after receiving them I'm happy to help.

mxndev, 2017-09-06:
But I tried to do some investigation. When I removed timeout option from my code, when I try to make subscription on characteristic, I wait forever and nothing happens nor I didn't get any error from your library in log console. What can i do more?

Kacper20, 2017-09-06:
You have `debug` operator which prints when subscription is made, when it's disposed, how many items are emitted. Try to check it and insert it in various places.

Kacper20, 2017-10-02:
Any advancements?

Kacper20, 2017-10-02:
I'm closing it for now — please reopen it when you have some questions. Please provide detailed description  & error logs so that we could in a reasonable time.

closed 2017-10-02 by Kacper20

hdmdmm, 2019-11-21:
Hello @Kacper20 
 I'm always getting the same error if I switched off my Bluetooth device. Yes, at subscribe I'm handling error:
>>Connection error has occured: The connection has timed out unexpectedly.<<
What to do if error is handled? Could you explain some experience of using CentralManager in  background.
If my device too far of bluetooth device the CentralManager might generate the error, but I would see it connecting/disconnecting.
Thank you,
DMitryKh.