Polidea/RxBluetoothKit #133: Send Data anyWhere

Xiahaiquan, 2017-08-16:
Hi 

    I  scanned  peripherals in my scanVC . then I want to send data to peripherals in other VC . how I can do . Like used on OC is singleton。so do ? or other way ?

Thank you

Kacper20, 2017-08-16:
Hello @Xiahaiquan 

Could you please elaborate more on your problem? To be honest I'm having troubles understanding your problem. Are you using example app? Could you provide some code sample?

Xiahaiquan, 2017-08-16:
OK ,        I mean is  I have a bluetoothManager in scanViewController. but I want use  the bluetoothManager  send data to peripherals at other ViewController . how can I do .              Thank you

Kacper20, 2017-08-21:
Hi @Xiahaiquan,

One thing you need to do is to pass `BluetoothManager` to the other controllers in initializers and that should be enough. Then, they'll have access to the manager, too.

Xiahaiquan, 2017-08-22:
OK @Kacper20 

I will try it.  Thank you

Kacper20, 2017-08-23:
Hello @Xiahaiquan,

Any progress on this?

Xiahaiquan, 2017-08-23:
@Kacper20 
   this has no more. Thank you

closed 2017-08-23 by Kacper20