Polidea/RxBluetoothKit #134: Does setNotificationAndMonitorUpdates work on macOS?

ctylim, 2017-08-21:
Hello, I have a problem with using `setNotificationAndMonitorUpdates` on **macOS**.
After connecting to peripheral and discovering services & characteristics (these process did work), I implemented 
```
// store discovered characteristics before this process
for characteristic in self.characteristics {
    characteristic.setNotificationAndMonitorUpdates()
        .observeOn(MainScheduler.instance)
        .subscribe(onNext: { characteristic in        
            print("notified: \(characteristic.value)")
        })
        .disposed(by: self.disposeBag)
}
```
I expected some notification messages appear in `.subscribe`, but this seems not worked.
Anything wrong or any similar problems on this? Here are my environments,

- macOS Sierra (10.12)
- Xcode 8.3.3
- Additional information: `Peripheral.setNotificationAndMonitorUpdates` also did not work.  `readValue` and `writeValue` worked fine.

Thank you in advance. Great project!

ctylim, 2017-08-21:
Ah, I resolved it by myself.
I tried to notify characteristics which does not have "notify" property (needs to be filtered).
Sorry for bothering you.

closed 2017-08-21 by ctylim

Kacper20, 2017-08-21:
Hi @ctylim,
Ahh, great! :-)
Happy to answer any other question.

Regards,

Kacper