Polidea/RxBluetoothKit #136: how to stop receive monitorValueUpdate from characteristic

sway0422, 2017-08-30:
I just want to stop receiving the updated value when my response data is received and done.

Kacper20, 2017-08-31:
Hi there,

Just call flatMap it with `setNotifyValue` with `false` parameter. In future versions of the library we're likely to make it easier to perform it using `Disposable`.

Regards,

Kacper

sway0422, 2017-08-31:
Thanks for your reply! 
Another question is about monitorValueUpdate(). I use this to handle updates coming from a characteristic. I am wondering if any way to check when the update is over. I also tried to onCompleted(), however, it's not been called. I thought it might be intentional. Thanks!

Kacper20, 2017-10-02:
Hi there,

Sorry for late response. As documentation specifies in generates `next`value with characteristic that was updated. It may fire multiple times. What do you mean by update is over? You mean that notifications has been disabled?

Kacper20, 2017-10-03:
So do I understand correctly that everything is fine since you've fixed it?
If yes — please close the issue.
If no — usually you do want to do these things in a serial way. This way you could identify specific requests.

closed 2017-10-11 by Kacper20