Polidea/RxBluetoothKit #137: How can I receive data per 0.2sec?

rosebaek, 2017-08-31:
I'd like to receive the data when I want.
If it is possible, I want to receive ther data per 0.2 sec.

Kindly check and give me the feedback.

Many thanks,
Jangmi

Kacper20, 2017-08-31:
Hi @rosebaek,

You cannot say that you want to receive new data from the device exactly per 0.2 sec. Actual response time from the device varies in time(based on the device speed itself, noise).

You could say that you want to execute read command every 0.2 sec and this could be done with `Observable.interval` operator.

rosebaek, 2017-08-31:
I don't know Observable.interval....
I'm sorry that detailed tell to me

Kacper20, 2017-08-31:
Here you could find the docs:
http://reactivex.io/documentation/operators/interval.html

Basically, you're giving it the time interval and it emits elements in given period after subscribe.
You could find some usages here: https://github.com/ReactiveX/RxSwift/blob/master/Documentation/GettingStarted.md

Kacper20, 2017-09-11:
Hi, any progress on this?

rosebaek, 2017-09-20:
no...I'm trying to solve this problem....
However,  I need to upgrade the rxBluetoothKit after installing ios11 and Xcode upgrade.

Kacper20, 2017-10-02:
If you have any further questions — please feel free to reopen the issue.

closed 2017-10-02 by Kacper20