Polidea/RxBluetoothKit #140: carthage update fail for 3.1

zhentan, 2017-09-07:
I'm getting the following errors when Carthage tries to build scheme "RxBluetoothKit iOS"
```
/RxBluetoothKit/Source/Descriptor.swift:27:33: error: Superfluous Disable Command Violation: SwiftLint rule 'line_length' did not trigger a violation in the disabled region. Please remove the disable command. (superfluous_disable_command)
Done linting! Found 3 violations, 1 serious in 29 files.
```
The offending line is 
`// swiftlint:disable line_length`
removing it seems to fix the problem. Also the "RxBluetoothKit macOS" scheme seems to build fine with or without this line.

Kacper20, 2017-09-11:
That's weird. Which version of swiftlint do you have installed on your system?

zhentan, 2017-09-11:
`swiftlint version` returned
`0.22.0`

qbraet, 2017-09-21:
same error for me, same swiftlint version

Kacper20, 2017-10-02:
It's fixed on newest release. Sorry for the issue.

closed 2017-10-02 by Kacper20