Polidea/RxBluetoothKit #141: scanForPeripherals not returning device after disconnection on iOS11

Ezor, 2017-09-13:
Hi,
 
Here is my workflow : 
1. `scanForPeripherals` : got `DeviceA`
2. `manager.connect(DeviceA)` 
3. working with device...
4.  ` manager.cancelPeripheralConnection(DeviceA)` 
5. `scanForPeripherals` : got **nothing** on iOS11, got `DeviceA ` on iOS10

Here is some snippet of my code, maybe i'm doing something wrong ?

```swift
func scan(_ delegate: BluetoothClientScanDelegate) {
        scanningDisposable = manager.rx_state
            .filter { $0 == .poweredOn }
            .take(1)
            .flatMap { _ in self.manager.scanForPeripherals(withServices: [BluetoothService.pocketPro.uuid], options:nil) }
            .subscribeOn(MainScheduler.instance)
            .subscribe(onNext: { device in
                delegate.onDeviceFound(device)
            }, onError: { error in
                delegate.onScanError(error)
            })
    }
```

```swift
func connectToDevice(_ device: Peripheral, delegate: BluetoothClientDeviceDelegate) {
        deviceDelegate = delegate
        currentDevice = device

        Logger.get().info("Connecting to device...")

        let _ = manager.connect(device).subscribe(
            onNext: { _ in
                Logger.get().info("Connected to device")
                self.monitorDisconnection()
                self.setupModeData()
            },
            onError: {
                Logger.get().error($0)
                self.deviceDelegate?.onDeviceFailed($0)
            }
        )
    }
```

```swift
func disconnectFromDevice() {
        deviceDelegate = nil
        disconnectionDisposable?.dispose()
        notificationDisposable?.dispose()
        guard let device = currentDevice else {
            return
        }
        currentDevice = nil
        _ = manager.cancelPeripheralConnection(device).subscribe()
    }
```

Kacper20, 2017-10-02:
Hi @Ezor,

Sorry for the late response. Which iOS version are you running? 
In terms of code everything looks just fine. One thing that is trivial: are you sure that when you were scanning for peripheral on iOS 11 device this peripheral was available? Wasn't it connected to another central? Could you check whether it's shown in the app like LightBlue on iOS device tell whether it's visible there?

Thanks for patience,

Kacper

Ezor, 2017-10-02:
Hi @Kacper20 

I was testing on iOS 11.0 GM

And yes the peripheral is available and not connected to another central as I can see it on my other iPhone (and Android device).
And I don't have this problem using a device running iOS10 with the same code.

Kacper20, 2017-10-02:
And are you able to see it on the same iOS device with the use of `LightBlue` app?

Kacper20, 2017-10-03:
I'm asking because I wonder what's the problem. In our library, we're doing  standard thing and it shouldn't break with iOS 11. Could you check whether `scanForPeripherals` is even called? You could do this by inserting `debug` operator after `take(1) and check for the logs in console.

Kacper

Ezor, 2017-10-03:
I will try all of this this week-end, thanks

Kacper20, 2017-10-11:
Hi there, any updates on this?

Ezor, 2017-10-11:
Hi,
Sorry I was too busy this week end, i will try to test it this evening

Kacper20, 2017-10-11:
Cool! :)

Ezor, 2017-10-11:
So, yes I can see the peripheral on Light Blue 

And here is the debug log (I got the same thing  on the first scan before I connect to the peripheral) : 
```
2017-10-11 18:12:42.741: BluetoothClient.swift:109 (scan) -> subscribed
2017-10-11 18:12:42.743: BluetoothClient.swift:109 (scan) -> Event next(poweredOn)
2017-10-11 18:12:42.744: BluetoothClient.swift:109 (scan) -> Event completed
2017-10-11 18:12:42.744: BluetoothClient.swift:109 (scan) -> isDisposed
```

I updated the library to 3.1.1 and my iPhone is running iOS 11.0.2.

Ezor, 2017-10-11:
I reduced the code of my app to do just this : 
1. `scanForPeripherals` : got `DeviceA`
2. `stopScan`
3. wait few seconds
4. `scanForPeripherals` : got nothing on iOS 11, got `DeviceA` on iOS 10

Here is the log on iOS 10 : 
```
[CoreBluetooth] API MISUSE: <CBCentralManager: 0x174273e40> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported
// SCANNING
2017-10-11 19:06:53.205: BluetoothClient.swift:113 (scan) -> subscribed
2017-10-11 19:06:53.206: BluetoothClient.swift:111 (scan) -> subscribed
2017-10-11 19:06:53.207: BluetoothClient.swift:109 (scan) -> subscribed
2017-10-11 19:06:53.217: BluetoothClient.swift:109 (scan) -> Event next(poweredOn)
2017-10-11 19:06:53.220: BluetoothClient.swift:109 (scan) -> Event completed
2017-10-11 19:06:53.220: BluetoothClient.swift:109 (scan) -> isDisposed
2017-10-11 19:06:53.263: BluetoothClient.swift:111 (scan) -> Event next(RxBluetoothKit.ScannedPeripheral)
2017-10-11 19:06:53.264: BluetoothClient.swift:113 (scan) -> Event next(RxBluetoothKit.ScannedPeripheral)
// STOPING SCAN
2017-10-11 19:06:57.149: BluetoothClient.swift:113 (scan) -> isDisposed
2017-10-11 19:06:57.151: BluetoothClient.swift:111 (scan) -> isDisposed
// SCANNING
2017-10-11 19:07:00.400: BluetoothClient.swift:113 (scan) -> subscribed
2017-10-11 19:07:00.402: BluetoothClient.swift:111 (scan) -> subscribed
2017-10-11 19:07:00.403: BluetoothClient.swift:109 (scan) -> subscribed
2017-10-11 19:07:00.404: BluetoothClient.swift:109 (scan) -> Event next(poweredOn)
2017-10-11 19:07:00.405: BluetoothClient.swift:109 (scan) -> Event completed
2017-10-11 19:07:00.405: BluetoothClient.swift:109 (scan) -> isDisposed
2017-10-11 19:07:00.576: BluetoothClient.swift:111 (scan) -> Event next(RxBluetoothKit.ScannedPeripheral)
2017-10-11 19:07:00.577: BluetoothClient.swift:113 (scan) -> Event next(RxBluetoothKit.ScannedPeripheral)
```

Here is the log on iOS 11 : 
```
2017-10-11 19:09:33.138169+0200 PocketPro[12217:3660334] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x1c027c800> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported
// SCANNING
2017-10-11 19:09:33.149: BluetoothClient.swift:113 (scan) -> subscribed
2017-10-11 19:09:33.150: BluetoothClient.swift:111 (scan) -> subscribed
2017-10-11 19:09:33.151: BluetoothClient.swift:109 (scan) -> subscribed
2017-10-11 19:09:33.158: BluetoothClient.swift:109 (scan) -> Event next(poweredOn)
2017-10-11 19:09:33.160: BluetoothClient.swift:109 (scan) -> Event completed
2017-10-11 19:09:33.160: BluetoothClient.swift:109 (scan) -> isDisposed
2017-10-11 19:09:33.165: BluetoothClient.swift:113 (scan) -> subscribed
2017-10-11 19:09:33.165: BluetoothClient.swift:111 (scan) -> subscribed
2017-10-11 19:09:33.166: BluetoothClient.swift:109 (scan) -> subscribed
2017-10-11 19:09:33.166: BluetoothClient.swift:109 (scan) -> Event next(poweredOn)
2017-10-11 19:09:33.166: BluetoothClient.swift:109 (scan) -> Event completed
2017-10-11 19:09:33.166: BluetoothClient.swift:109 (scan) -> isDisposed
2017-10-11 19:09:33.201: BluetoothClient.swift:111 (scan) -> Event next(RxBluetoothKit.ScannedPeripheral)
2017-10-11 19:09:33.201: BluetoothClient.swift:113 (scan) -> Event next(RxBluetoothKit.ScannedPeripheral)
2017-10-11 19:09:33.203: BluetoothClient.swift:111 (scan) -> Event next(RxBluetoothKit.ScannedPeripheral)
2017-10-11 19:09:33.203: BluetoothClient.swift:113 (scan) -> Event next(RxBluetoothKit.ScannedPeripheral)
// STOPING SCAN
2017-10-11 19:09:37.775: BluetoothClient.swift:113 (scan) -> isDisposed
2017-10-11 19:09:37.775: BluetoothClient.swift:111 (scan) -> isDisposed
// SCANNING
2017-10-11 19:09:39.180: BluetoothClient.swift:113 (scan) -> subscribed
2017-10-11 19:09:39.181: BluetoothClient.swift:111 (scan) -> subscribed
2017-10-11 19:09:39.181: BluetoothClient.swift:109 (scan) -> subscribed
2017-10-11 19:09:39.181: BluetoothClient.swift:109 (scan) -> Event next(poweredOn)
2017-10-11 19:09:39.181: BluetoothClient.swift:109 (scan) -> Event completed
2017-10-11 19:09:39.181: BluetoothClient.swift:109 (scan) -> isDisposed
```

Ezor, 2017-10-11:
I found the problem....

I was calling scanForPeripheral twice (without stopping scan between each call).

I got the following code in my main controller : 

```swift
override func viewDidAppear(_ animated: Bool) {
        super.viewDidAppear(animated)

        scan()

        NotificationCenter.default.addObserver(self, selector: #selector(appIsGoingForeground), name: .UIApplicationDidBecomeActive, object: nil)
        NotificationCenter.default.addObserver(self, selector: #selector(appIsGoingBackground), name: .UIApplicationWillResignActive, object: nil)
    }

@objc func appIsGoingForeground() {
        scan()
    }

func scan() {
 // scanForPeripheral...
}
```

As you can see I start the scan as soon as my controller appears and I register observers to be able to stop/restart scan when the app goes background/foreground.

On iOS 11 a `UIApplicationDidBecomeActive` notification is sent after my controller appeared and I was starting a new scan.
On iOS 10 (or maybe because my device is quite slow) I never receive the `UIApplicationDidBecomeActive` notification, so no problem

Kacper20, 2017-10-12:
To ensure that there is only one scan at the moment you could use `SerialDisposable`. 
Do you have any other questions related to the library?

Ezor, 2017-10-12:
No that’s OK, many thanks for your help

closed 2017-10-12 by Ezor