Polidea/RxBluetoothKit #144: Better Bluetooth Resetting

MarvinNazari, 2017-09-25:
So CoreBluetooth sometimes throws bluetooth resetting which is basically `BTServer` crashing inside `iOS` right now RxBluetoothKit throws the `bluetoothResetting`.

You will get `CoreBluetooth[ERROR] XPC connection interrupted, resetting.` in console.

If I'm not wrong the easiest way to handle this correctly is that you need to remove all the peripheral, services, characteristics that `BluetoothManager` have stored and reinit `CBCentralManager` and basically start all over.

Im not saying that RxBluetoothKit should handle this, but a function in `BluetoothManager` which u are enable to reset the stuff above will be really helpful, I'm interested to know how u guys handle this issue as well.

Kacper20, 2017-10-02:
Will think about it. Problem is that our `CBCentralManager` instance is injected from the outside. This makes things complicated and would definitely require breaking the API (because instead object we should take a closure that returns object and use it at start and every time when bluetooth is reseting).

Kacper20, 2017-10-26:
We think that you should just re-initialize the `BluetoothManager` if anything like that happens. Because we're injecting `CBCentralManager` we don't see a way to implement this in a clear way. 
Doing more operations is outside of the library scope. Also it has it's benefits — we're ensuring that all of the state internal object is cleared just because it's deallocated. It makes us certain that we don't need to remember about clearing whenever we're adding any internal storage to the `BluetoothManager`.

If you have some further thoughts — feel free to re-open the issue. I'm open to the discussion about this feature.

closed 2017-10-26 by Kacper20