Polidea/RxBluetoothKit #149: Can not run RxBluetoothKit (Swift 4.0) on iPhone 5c (10.3 iOS)

rgarciaotc, 2017-10-03:
I am trying to run the example with my own device and can not do it because at the time that I choose my own device i can not acceded to any method of the library, if I jump, I just can see autogenerated code by xcode. But if I select a simulator device i can Jump to the definition and see all the implemented methods.
[image]
[image]
[image]

Kacper20, 2017-10-03:
We haven't updated the example for Swift 4.0 yet. I will do this today and make sure that all works as expected.

rgarciaotc, 2017-10-03:
Really thanks I guess can find it at swift 4.0 branch.

Kacper20, 2017-10-03:
Yes, will be available in a couple of hours. Thanks for the patience!

rgarciaotc, 2017-10-03:
I know you are making changes, but I can not understand why when I try to build I can not find any method from RxBluetoothKit, why xcode autogenerated some code instead of let me see the implemented methods.

Kacper20, 2017-10-04:
I'm not sure why it's like that. Right now working example is available on the `swift4` branch. Please run `pod repo update` and `pod install` to make it running :)

closed 2017-10-04 by Kacper20