Polidea/RxBluetoothKit #151: Carthage update error in macOS Sierra & Xcode 9

ctylim, 2017-10-04:
Hi, I have a trouble with Carthage build.

Environments:

- Xcode 9.0
- Swift 3.2
- macOS 10.12 Sierra

I did:

```
$ carthage update --platform macos --no-use-binaries
```

Then console said:

```
*** Fetching RxSwift
*** Checking out RxBluetoothKit at "3.1.1"
*** Checking out RxSwift at "3.6.1"
*** xcodebuild output can be found in /var/folders/qr/55wc70594lbb9xnq6352rb880000gn/T/carthage-xcodebuild.0he5VY.log
*** Building scheme "RxTests-macOS" in Rx.xcworkspace
*** Building scheme "RxSwift-macOS" in Rx.xcworkspace
*** Building scheme "RxBlocking-macOS" in Rx.xcworkspace
*** Building scheme "RxCocoa-macOS" in Rx.xcworkspace
*** Building scheme "RxBluetoothKit macOS" in RxBluetoothKit.xcodeproj
Build Failed
	Task failed with exit code 65:
	/usr/bin/xcrun xcodebuild -project [***]/Carthage/Checkouts/RxBluetoothKit/RxBluetoothKit.xcodeproj -scheme RxBluetoothKit\ macOS -configuration Release -derivedDataPath [***]/Library/Caches/org.carthage.CarthageKit/DerivedData/9.0_9A235/RxBluetoothKit/3.1.1 ONLY_ACTIVE_ARCH=NO CODE_SIGNING_REQUIRED=NO CODE_SIGN_IDENTITY= CARTHAGE=YES build (launched in [***]/Carthage/Checkouts/RxBluetoothKit)

This usually indicates that project itself failed to compile. Please check the xcodebuild log for more details: /var/folders/qr/55wc70594lbb9xnq6352rb880000gn/T/carthage-xcodebuild.0he5VY.log
```

Error messages in log file:

```
[***]/Carthage/Checkouts/RxBluetoothKit/Source/Logging.swift:155:36: error: 'identifier' is only available on OS X application extension 10.13 or newer
        return "Peripheral(uuid: \(identifier), name: \(String(describing: name)))"
                                   ^
[***]/Carthage/Checkouts/RxBluetoothKit/Source/Logging.swift:154:9: note: add @available attribute to enclosing var
    var logDescription: String {
        ^
[***]/Carthage/Checkouts/RxBluetoothKit/Source/Logging.swift:153:1: note: add @available attribute to enclosing extension
extension CBPeripheral : Loggable {
^
[***]/Carthage/Checkouts/RxBluetoothKit/Source/RxCBPeripheral.swift:47:27: error: 'identifier' is only available on OS X application extension 10.13 or newer
        return peripheral.identifier
                          ^
[***]/Carthage/Checkouts/RxBluetoothKit/Source/RxCBPeripheral.swift:47:27: note: add 'if #available' version check
        return peripheral.identifier
                          ^
[***]/Carthage/Checkouts/RxBluetoothKit/Source/RxCBPeripheral.swift:46:9: note: add @available attribute to enclosing var
    var identifier: UUID {
        ^
[***]/Carthage/Checkouts/RxBluetoothKit/Source/RxCBPeripheral.swift:31:7: note: add @available attribute to enclosing class
class RxCBPeripheral: RxPeripheralType {
      ^

** BUILD FAILED **

The following build commands failed:
	CompileSwift normal x86_64
	CompileSwiftSources normal x86_64 com.apple.xcode.tools.swift.compiler
(2 failures)

```

Does anyone know how to resolve this compile error?

ctylim, 2017-10-04:
I guess this error derives from API update in CoreBluetooth, so I manually changed Deployment Target to 10.13 in Carthage/Checkouts/RxBluetoothKit/RxBluetoothKit.xcodeproj.

Cierpliwy, 2017-10-04:
Unfortunately there is a breaking change in CoreBluetooth API and transition isn't straightforward. I'm working on it on branch `ver_0_7`. You can try it out if it helps.

Kacper20, 2017-10-04:
Hi @ctylim,

Please use the `4.0.0-alpha.1` tag :) This one supports Swift 3.2 & Swift 4.
We'll have a stable version once RxSwift releases one.

Kacper20, 2017-10-04:
I'm updating the README right now to let users know about it.

ctylim, 2017-10-04:
Oh, I thought the 4.x versions are compatible with only Swift 4. And it worked!
Thank you for your reply!

Kacper20, 2017-10-04:
👍 We'll release a stable version once our dependency (RxSwift) is stable. Thanks for the feedback!

closed 2017-10-04 by Kacper20