Polidea/RxBluetoothKit #154: Merging notification and read

appoulsen, 2017-10-11:
I'm getting a crash when merging notification and read.
I had no problem at version 3.1.

```
        let monitorSettings = peripheral.setNotificationAndMonitorUpdates(for: CABLECharacteristic.settings).map{$0.value}.filterNil()
        let listOfCmds = Observable.from(["BA", "BR", "DB", "MN", "VL"]).map{$0.data(using: String.Encoding.ascii)}
        let readSettings = listOfCmds.filterNil().flatMap{self.peripheral.writeValue($0, for: CABLECharacteristic.settings, type: .withResponse)}
            .flatMap{$0.readValue()}.map{$0.value}.filterNil()
        let merged = Observable.merge(monitorSettings, readSettings)
```

Kacper20, 2017-10-12:
Could you tell what kind of crash is it? Please provide the error log & backtrace

appoulsen, 2017-10-12:
[image]
[image]

2017-10-12 14:18:46.511592+0100 CAInterfaceApp[2167:960954] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x1c0463b40> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported
2017-10-12 14:18:46.562743+0100 CAInterfaceApp[2167:960954] refreshPreferences: HangTracerEnabled: 0
2017-10-12 14:18:46.562784+0100 CAInterfaceApp[2167:960954] refreshPreferences: HangTracerDuration: 500
2017-10-12 14:18:46.562801+0100 CAInterfaceApp[2167:960954] refreshPreferences: ActivationLoggingEnabled: 0 ActivationLoggingTaskedOffByDA:0

Kacper20, 2017-10-12:
And could you show the line from the stack 6 frame? `specialized Peripheral.setNotification`
Thanks :)

Kacper20, 2017-10-19:
Hi @appoulsen, any updates?

appoulsen, 2017-10-19:
after I rebuilt with my own compiler I haven't been able to reproduce the problem. On the other hand I'm having problems with sending multiple values to a single characteristic as I received different values. but Im digging into it. ill post a issue when I know a bit more

closed 2017-10-19 by appoulsen