Polidea/RxBluetoothKit #155: How to get names of peripherals?

keeninter, 2017-10-11:
I ran the Example and it always show the UUIDs of all BLE devices in range, not their names. It seems that localname in
`peripheral.advertisementData.localName ?? peripheral.peripheral.identifier.uuidString`
was always nil. I tried it in my code also and peripheral.advertisementData.localName and peripheral.peripheral.name had been indeed nil.
What is the best approach to get the names of peripherals in range advertising this info?

Kacper20, 2017-10-12:
Hi there!
Not every peripheral is advertising it's local name.  When I'm running it on my device there are couple of named ones, so the approach is definitely good — you just need to find the proper device :)

Kacper20, 2017-10-12:
I'm closing for now — please feel free to reopen if you have further questions.

closed 2017-10-12 by Kacper20

keeninter, 2017-10-12:
Hi,
I've made an interesting -- for me, at least -- observation that after LightBlue application connects with any of my devices and briefly disconnects, I launch my code based on the Example and actually get that peripheral's name. It looks like a device can be "urged" to advertise its local name. I can speculate that (iOS) apps like Bluefruit or BluetoothLE which more successful in getting local names of peripherals and macOS which shows names of all my devices in Bluetooth Preferences behind the scenes somehow urge devices in range to advertise their names.

Kacper20, 2017-10-12:
Local name can be read from the characteristic on the device. iOS is caching these values and after you scan with example app — the id discovered by connecting via the LightBlue is returned.