Polidea/RxBluetoothKit #156: BluetoothManager deinit

taraskontsur, 2017-10-12:
Hi, guys. There are retain cycle issues in `BluetoothManager`. For example:
```
    let disposeBag = DisposeBag()
    var manager: BluetoothManager?
    
    @IBAction func action(_ sender: UIButton) {
        if manager == nil {
            manager = BluetoothManager()
            manager?.rx_state.filter { $0 == .poweredOn }.debug().subscribe().disposed(by: disposeBag)
        } else {
            manager = nil
        }
    }
```
Steps:
- tap to the button and turn bluetooth on
```
2017-10-12 16:13:50.482: TestViewController.swift:21 (action) -> subscribed
2017-10-12 16:13:50.482: TestViewController.swift:21 (action) -> Event next(poweredOn)
```
- tap again, now `manager = nil` but deinit never called
- tap again, create new manager instance
```
2017-10-12 16:19:11.940: TestViewController.swift:21 (action) -> subscribed
2017-10-12 16:19:11.956: TestViewController.swift:21 (action) -> Event next(poweredOn)
```
- turn bluetooth off and on, `(action) -> Event next(poweredOn)` print twice which is not ok
```
2017-10-12 16:21:16.438: TestViewController.swift:21 (action) -> Event next(poweredOn)
2017-10-12 16:21:16.438: TestViewController.swift:21 (action) -> Event next(poweredOn)
```

Proposed solution
https://github.com/Polidea/RxBluetoothKit/blob/master/Source/BluetoothManager.swift#L211
```
    public var rx_state: Observable<BluetoothState> {
        return .deferred { [unowned self] in
            return self.centralManager.rx_didUpdateState.startWith(self.centralManager.state)
        }
    }
```
or
```
    public var rx_state: Observable<BluetoothState> {
        return .deferred { [central = self.centralManager] in
            return central.rx_didUpdateState.startWith(central.state)
        }
    }
```

Kacper20, 2017-10-12:
You're right. I will release a fix on the weekend/beginning of next week.
Thanks for the feedback!

taraskontsur, 2017-10-12:
@Kacper20 thanks, don't forget to fix other operators which keeps strong references)

Kacper20, 2017-10-20:
Fixed by #158, will be released in 4.0 version

PR #158 (2017-10-20): 4.0 release, memory leaks fixes
Fixes #156 , final preparation for official 4.0 release
closed 2017-10-20 by Kacper20