Polidea/RxBluetoothKit #157: Build errors for macOS lower than 10.13

xtazz, 2017-10-18:
When building latest release version in XCode 9.1 with Swift 3.2, it gives build errors because library uses **identifier** field in **CBPeer** which is available only [starting from 10.13](https://developer.apple.com/documentation/corebluetooth/cbpeer/1620687-identifier), which is definitely not true. Need some workaround for this, seems like this one should work: https://forums.developer.apple.com/thread/84375

Kacper20, 2017-10-18:
Could you try the `4.0-rc` tag? It should work with Swift 3.2. I will release `4.0` in a moment.

xtazz, 2017-10-18:
It's not about Swift version, but more like the macOS SDK version. And I don't see any related change to 'identifier' field for peripheral in 4.0 branch. 

For now I have a workaround like this, but I believe that there is some better way to fix it:
`
extension CBPeripheral {
    override open var identifier: UUID {
        get {
            if #available(macOS 10.13, *) {
                return super.identifier
            } else {
                return value(forKey: "identifier") as! NSUUID as UUID
            }
        }

        set {
            setValue(newValue, forKey: "identifier")
        }
    }
}
`

Kacper20, 2017-10-18:
I'm aware of your question. It builts with the newest macOS SDK, as I've said. Please check 4.0 branch :)

xtazz, 2017-10-18:
@Kacper20 good 👍 I can't try it now cause I don't want to spend time transitioning the project to RxSwift 4.x. It would be good to port your fix to 3.x

Kacper20, 2017-10-18:
Oh, okay. Will do it tomorrow :)

Kacper20, 2017-10-24:
Hi @xtazz,
Sorry for the delay. It's finally released under `3.1.2` version. Enjoy :)

closed 2017-10-24 by Kacper20