Polidea/RxBluetoothKit #161: After connecting and discovering a chracteristic setnotificationandmonitorupdates fires a change error

Neogene, 2017-10-21:
Hi, after I detect all the bt devices I want to liste to a specific message, docs say:

 Servcie Main UUID:0xFFE0  Sub UUID：0xFFE1 prop= CHAR_PROP_READ | CHAR_PROP_NOTIFY     The command format is as follows Byte0 0x00 Each click a button, to send one time data; click on camera interface to take photos;  Positioning on other interface; Dual click trigger alarm or record  

Using this code sends after the setNotificationAndMonitorUpdates the error: "Characteristic notify change error has occured: The attribute could not be found."

Any help? Maybe I'm missing some additional chaining. best regards.

`     peripheral.connect()
            .flatMap { $0.discoverServices([CBUUID.init(string: "FFE0")]) }
            .flatMap { Observable.from($0) }
            .flatMap { $0.discoverCharacteristics([CBUUID.init(string: "FFE1")])}
            .flatMap { Observable.from($0) }
            .subscribe(onNext: { characteristic in
                print("Discovered characteristic: \(characteristic.uuid.uuidString)")
                characteristic.monitorWrite()
                    .subscribe(onNext: { characteristic in
                        print("write detected \(characteristic)");
                    })
                characteristic.monitorValueUpdate()
                    .subscribe(onNext: {characteristic in
                        print("update detected \(characteristic)");
                    })
                characteristic.setNotificationAndMonitorUpdates()
                    .subscribe(onNext :{ characteristic in
                        print("subs: \(characteristic)")
                    })
                
            }
          )`

Kacper20, 2017-10-25:
Hi there,

After reading your questions I assume that you have two questions.
## BT State issue
It really depends on your logic. You could just wrap all your logic into a function. Inside onError you could check for a timeout error. Then just call this function that subscribes to state again. It all depends on your logic — do you want to show the error to the user? If not you could omit the timeout operator. If you want to have some specific behaviour — please let me know and I'll give you a hints.

## Reconnection
To reconnect after you establish the connection you should begin subscription to `monitorPeripheralDisconnection`. After event that is informing about the disconnection is emitted you should try connect again to the peripheral.

Kacper20, 2017-10-26:
Hi @Neogene,
Any updates? Does it answers your concerns?

Kacper20, 2017-10-30:
Closing as of now. Please feel free to reopen the issue if you have any additional questions! :)

closed 2017-10-30 by Kacper20