Polidea/RxBluetoothKit #164: [5.0] Remove reference to deprecated OSAtomicCompareAndSwap32

Kacper20, 2017-10-22:
Inside our library we are using function which was deprecated in iOS 10 && macOS 12 SDKs. It’s referenced [here](https://github.com/Polidea/RxBluetoothKit/blob/f9cd5f0bbc191f0ba49e9fb65845518b1effea62/Source/Observable%2BQueueSubscribeOn.swift#L143). We should remove the reference to this function and use the analogous function from `stdatomic.h`

# How it affects current users
There is no affection.

siejkowski, 2017-10-26:
# Discussion needed

When doing this tak, I've encountered the problem with using functions from `stdatomic.h` directly in Swift.

## Problem

The issue is that although `stdatomic.h` is public under current SDKs and it's added to module map of `Darwin.C` module,  the `import Darwin.C.stdatomic` doesn't expose required `atomic_compare_exchange_strong_explicit` function. It's a known bug: https://openradar.appspot.com/27161329 but it hasn't been fixed at the moment.

## Possible solutions

There are three possible solutions I've found:

* add `stdatomic.h` to module map of `RxBluetoothKit`. I can be done either directly or using a wrapping module that makes the Swift-friendly adjustments to the `stdatomic` interface. This is the approach that [swift-atomic](https://github.com/glessard/swift-atomics) library takes, see [here](https://github.com/glessard/swift-atomics/tree/master/Xcode/Atomics)

* write our own implementation of atomic compare and swap, similar to what [RxSwift has done on Linux](https://github.com/ReactiveX/RxSwift/blob/d52c3a8628e86183879f7852e34f0437e7b44836/Platform/Platform.Linux.swift#L71)

* ignore the deprecation warning and continue using `OSAtomicCompareAndSwap32`. This is the approach that [RxSwift took on Darwin platforms](https://github.com/ReactiveX/RxSwift/blob/master/Platform/Platform.Darwin.swift)

## Discussion

The first solution is following the Apple hint, so it might be the clearest. However, it's complicating the project structure and possibly might break in the future, if the availability of `stdatomic.h` would change in next SDK release.

The second solution is non-invasive, but least optimised. It needs performance consideration.

The third solution is a non-solution, and it might also break in the future, when deprecation will turn into lack of availability of `Darwin.libkern.OSAtomic`.

@Kacper20 @Cierpliwy What do you think?

Kacper20, 2017-10-26:
Hi @siejkowski,

Thanks so much for your analysis — it's very thorough! We've been talking about getting rid of the complicated scanning logic as it's pain to maintain and isn't really usable. 
Because our custom operator & scheduling is used only in scanning and it's also the class in which we see this deprecation we'll most surely remove all of the related code. Based on this I'm closing this issue. If we'll change our minds we'll come back to this topic and it'll be reopened. It's also a great resource for other developers having the same problems.

I will make an issue soon about the details of scan refactor.

closed 2017-10-26 by Kacper20