Polidea/RxBluetoothKit #166: [5.0] Support for CBPeripheralManager

Kacper20, 2017-10-23:
Our library currently only supports `CoreBluetooth` central mode. In the 5.0 versions we could like to support API that allows the Apple device to become peripheral. 
Since the naming of two managers in `CoreBluetooth` is as follows: `CBCentralManager`, `CBPeripheralManager` — it would make sense to rename our current `BluetoothManager` to `CentralManager`. New object would then be named `PeripheralManager`.

# Design
Since `CBCentralManager` and `CBPeripheralManager` both inherit from `CBManager` which consist `state` property it would be appropriate to map the same relationship in our hierarchy in order to avoid code duplication. It could be named `BaseManager` and should take `CBManager` instance in initializer. This class would expose functions that operate on state, that currently belong to the `BluetoothManager`, like continuous access to `BluetoothState`.
Then, both new `CentralManager` and `PeripheralManager` would inherit from `BaseManager`.

Design of `PeripheralManager` should be analogous to the current `BluetoothManager` (After changes from #162 . One that implements it should perform pairing of methods from `CBPeripheralManager` and the delegates callbacks. Then, every method should be wrapped in an reactive way. Delegate proxy for the `CBPeripheralManagerDelegate` should also be created.

# Affection to current users
Rename of main user facing object is a source breaking change. We could make it easier to the users by introducing fixit directive. All other changes are purely additive.

paweljaneczek, 2018-05-22:
@danielbotka hey, we are on the final steps before releasing RC version with `PeripheralManager` support. 
Problem here is that currently we are not developing any project that will use that feature. Would you like to test it before releasing stable version?

danielbotka, 2018-05-22:
@paweljaneczek hey, yes, I'd be happy to play around a little, probably next week. Is it on a specific branch?

paweljaneczek, 2018-05-22:
@danielbotka great to hear! In next week we will have probably RC library version so we will give a sign to you in this thread

jodm, 2018-05-23:
@paweljaneczek We also have a project that we could integrate this into. Should I try and integrate from the `feature/peripheral_manager` branch?

paweljaneczek, 2018-05-24:
@jodm great! For now the newest changes are on `feature/peripheral_manager_tests`

paweljaneczek, 2018-05-28:
@jodm @danielbotka We've just release version 5.1.0 with support for `PeripheralManager`. If you will find any bugs/problems for it just fill new issue on github. Closing this one.

closed 2018-05-28 by paweljaneczek