Polidea/RxBluetoothKit #172: Problem in Core Bluetooth API that is present also in RxBluetoothKit

vladiulianbogdan, 2017-10-24:
I've forked your repository and I've started working on adding the RX PeripheralManager functionality. I've encountered a weird thing in the Core Bluetooth API that is present also in the RxBluetoothKit.

Let's take CBCharacteristic as an example. CBMutableCharacteristic inherits from CBCharacteristic. The problem is that CBCharacteristic has a non optional property of type CBService and when I create a CBMutableCharacteristic I don't have a service for it.

When the delegate method below gets called in Core Bluetooth:

`optional func peripheralManager(_ peripheral: CBPeripheralManager, didAdd service: CBService, error: Error?)`

the service is actually a CBMutableService and its peripheral property is nil. If you try to access it, it will crash.

The RxBluetoothKit has the same issue: if I want to create a Characteristic, I have to pass a Service to it, but in the universe of PeripheralManager I don't have a Service to associated with it. The compiler will not allow me to create a Characteristic (or MutableCharacteristic that inherits from Characteristic) without providing a Service in the constructor.

Two possible solutions for this would be:

1. Make these properties implicitly unwrapped and behave the same way as Core Bluetooth.
2. Make these properties optional and returns an Observable.error if someone access them incorrectly.

Kacper20, 2017-10-26:
Hi @vladiulianbogdan,

Just to let you know — I'm getting to this place in my implementation and we'll let you know what I think soon. Thanks for the patience.

vladiulianbogdan, 2018-03-12:
Hi! Any status about the new version which includes PeripheralManager feature?

paweljaneczek, 2018-03-13:
Hey @vladiulianbogdan 
We are finishing version 5.0 which will be release very soon. After it we will start working on `PeripheralManager` feature

paweljaneczek, 2018-03-13:
Hmm there is already ticket #166, so closing this for now

closed 2018-03-13 by paweljaneczek