Polidea/RxBluetoothKit #183: [5.0] Incorrect error check in L2CAP channel open

tomipu, 2017-11-21:
When opening L2CAP channel, error callback is always executed. Problem is most likely in the error check that should be `error == nil` instead of `error != nil`.

Code snippet from Peripheral.swift:
```
let observable = delegateWrapper.rx_didOpenL2CAPChannel
    .take(1)
    .flatMap { [weak self] (channel, error) -> Observable<CBL2CAPChannel> in
        guard let strongSelf = self else { throw BluetoothError.destroyed }
        if let channel = channel, error != nil {
            return .just(channel)
        } else {
            throw BluetoothError.openingL2CAPChannelFailed(strongSelf, error)
        }
}
```

Kacper20, 2017-11-21:
Hi @tomipu, you’re entirely right. Thanks so much!

@scibas could you send a patch to this?

Kacper20, 2017-11-21:
It’s fixed by #184. 

Thanks again for feedback!

closed 2017-11-21 by Kacper20

tomipu, 2017-11-22:
Wow, that was fast!
Thanks guys, works like charm now.