Polidea/RxBluetoothKit #185: What is `serviceIds`

rohitnewscorp, 2017-11-28:
I just started this library. It is well-explained library except for one thing i.e  What is `serviceIds `? and how can I create it? Kindly help as I need to integrate this library as soon as possible.

Thanks & regards

bartoszstelmaszuk, 2017-12-01:
What exactly `serviceIds` you are mentioning about? 

Is it `ServiceIdentifier` protocol defined in `DeviceIdentifiers` file? If it is than this convenience protocol allows you to represent your services handles as enum cases and makes it easier to get discovered services.
You can read service identifiers from manufacturer's documentation or by scanning and exploring your bluetooth device.

paweljaneczek, 2018-01-15:
Closing due to no answer. @rohitnewscorp If you have any questions please open new ticket

closed 2018-01-15 by paweljaneczek