Polidea/RxBluetoothKit #190: Not able to find peripheral name

rohitjb, 2017-12-03:
Hi 

I am working on an app where I need to show the name of devices in a list which are in the bluetooth range so I am using this library which is really cool but the issue is I am not able to find the name of peripheral. Kindly help me with this.

Thanks & regards

bartoszstelmaszuk, 2017-12-04:
Can you give some more information about how you try to find the name of peripheral on app side? Is it problem to discover a device or you discover device but couldn't read the name from data?

Most importantly tell more about device. Does it transmit proper `advertisementData` so it could be discovered? iOS tends to cache `localName` so maybe app at the beginning receives different name than you suspect and returns it later from cache?

paweljaneczek, 2018-01-09:
@rohitjb do you still have a problem or is it already resolved?

paweljaneczek, 2018-01-15:
Closing due to no answer. @rohitjb If you have any questions please open new ticket

closed 2018-01-15 by paweljaneczek