Polidea/RxBluetoothKit #192: Triggering re-scan turns off scanning

mwoollard, 2017-12-08:
Take the following intending to re-scan for peripherals every 5 minutes. Only the first scan request works. Debugging shows that the existing scan observable is picked for re-use and then that observable's dispose closure fires afterwards removing the observable wrapper from the queue and issuing a stop scan call to CB manager leaving an observable that will never emit any events.

```
        manager.rx_state
            .filter { $0 == .poweredOn }
            .take(1)
            .flatMap { _ in Observable<Int>.timer(0, period: 300, scheduler: MainScheduler.instance) }
            .flatMapLatest { _ in manager.scanForPeripherals(withServices: nil) }
            .subscribe()
```

Cierpliwy, 2017-12-08:
Did you test version 4.0.1?

paweljaneczek, 2018-01-15:
Closing due to no answer. @mwoollard If you have any questions please open new ticket

closed 2018-01-15 by paweljaneczek