Polidea/RxBluetoothKit #194: re-connect or wake-up bluetooth device when app is terminated

yusufonderd, 2017-12-13:
I can discover all characteristic and services in foreground. I listen notificationUpdates from spesific characteristic. When notify characteristic,  My app shows local notification. There is no problem in foreground and background. But when app is terminated, ble device is disconnected. So I want to re-connect again ble device or preserve connection when app is terminated . How can I do this ?  I'm using restoreKeyIdentifier for this : 

//restore identifier 
` var restoreOptions = [CBPeripheralManagerOptionRestoreIdentifierKey: "rxRestoreIdentifier"] as [String : AnyObject]`
//to connect peripheral and subscribe notification characteristic 
```
peripheral.connect(options: restoreOptions)
...
.subscribe(onNext: { characteristic in
characteristic.setNotificationAndMonitorUpdates()
                            .subscribe(onNext :{ characteristic in
                                     showNotification()
                            }})
}
)
```
// I'm using listenOnRestoredState in didFinishLaunchingWithOptions 
```
manager.listenOnRestoredState()
            .subscribe(onNext: { (restoredState) in
                
            },onError: {error in
                print(error)
            },onCompleted:{
                print("asdf")
            },onDisposed:{
                print("monitorDisconnection onDisposed" )
            })

```
But it doesn't work . Where is my mistake ?

nhatlee, 2017-12-18:
I faced the same problem here. After the app terminated, I also want to keep connections with MIDI device. How to do that?

paweljaneczek, 2018-01-15:
@nhatlee @yusufonderd How are you testing app termination? If you are doing force quit or closing app from multitask panel then system `thinks` that user don't want to use this app anymore. It means that system won't listen for ble device specified in your app and won't open your app in background.

nhatlee, 2018-01-16:
@paweljaneczek : I solved the problem by call `MIDIRestart` after connect successful to BLE.

paweljaneczek, 2018-01-19:
@nhatlee great!
@yusufonderd is this issue still actual?

yusufonderd, 2018-01-22:
Hi problem fixed.

closed 2018-01-24 by paweljaneczek

thuydao, 2020-02-10:
> @paweljaneczek : I solved the problem by call `MIDIRestart` after connect successful to BLE.

I have same problem, can you explain to detail, many thanks