Polidea/RxBluetoothKit #195: Unable to infer complex closure return type; add explicit type to disambiguate

vitorventurin, 2017-12-19:
When I try:

manager.scanForPeripherals(withServices: [servicesId])
      .flatMap { scannedPeripheral in //error line
        let advertisement = scannedPeripheral.advertisement
    }

I receive an error message: Unable to infer complex closure return type; add explicit type to disambiguate

DevAndArtist, 2017-12-21:
This is Swift not able to infer the type for you and due the complexity of `RxSwift`. Do as the compiler error says and provide explicit type like `(scannedPeripheral: Peripheral) -> Observable<Peripheral>`.

But to be honest, this is simply a wrong example from the README. `.flatMap` simply makes not sense for that example. Exchange it with `.subscribe(onNext: { scannedPeripheral in })`.

paweljaneczek, 2018-01-15:
Changed readme in #203. Closing.

closed 2018-01-15 by paweljaneczek