Polidea/RxBluetoothKit #197: API MISUSE: CBCentralManager has no restore identifier

leandromperez, 2017-12-21:
Hello guys, I don't know if this is the proper place to ask this, but I'm seeing this Warning in my console

`[CoreBluetooth] API MISUSE: <CBCentralManager: 0x1c047e500> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported`

I made sure I am only making 1 call to scanForPeripherals. Any thoughts?

DevAndArtist, 2017-12-21:
This is how I initialize a singleton `manager` to get rid of the issue:

```

leandromperez, 2017-12-26:
@DevAndArtist Thanks for your hint. I am seeing that warning from time to time but definitely not as often as before. I have another question that might be related to the initialization as well.

How do I ask the user to turn on bluetooth?

This is how I'm instantiating the manager now.
let restoreKey = CBCentralManagerOptionRestoreIdentifierKey
let showAlertKey = CBCentralManagerOptionShowPowerAlertKey
self.manager = BluetoothManager(queue: .main, options: [showAlertKey : true as AnyObject, restoreKey : "my identifier" as AnyObject])

I am not seeing any UIAlert though. So, when I start scanning for peripherals I get an error: .bluetoothPoweredOff 

Any ideas?

Sorry if this is the wrong place for asking this, perhaps gitter or stackoverflow are the proper places?

Thanks a million

paweljaneczek, 2018-01-15:
Problem solved on gitter. Answer:

> In iOS 11 turning it off from control center doesn't fully disables bluetooth - in a result we have situation when this alert is not shown and we get bluetoothPoweredOff. When you would turn it off in settings -> bluetooth then you should get UIAlertView.
> The only solution here is to show your own UIAlertView.

closed 2018-01-15 by paweljaneczek