Polidea/RxBluetoothKit #198: How can I get the nearest peripheral (largest RSSI) ?

Anccerson, 2017-12-23:
Currently I get the target peripheral by detecting whether its RSSI is larger than -40. This is my code:
```swift
func scan(targetId: String? = nil, scanOptions: [String: Any]? = nil) -> Observable<Peripheral> {
    return bluetoothManager.rx_state.filter({ state in
        if state == .poweredOn {
            return true
        } else if state == .poweredOff {
            throw BluetoothError.bluetoothPoweredOff
        } else {
            return false
        }
    }).flatMap { _ in
        return self.bluetoothManager.scanForPeripherals(withServices: [DeviceService.huffmanCode.uuid], options: scanOptions)
    }.filter { peripheral in
        if let targetId = targetId {
            return UUID(uuidString: targetId) == peripheral.peripheral.identifier
        } else {
            return peripheral.rssi.intValue >= -40
        }
    }.take(1).map { scannedPeripheral -> Peripheral in
        self.casePeripheral = scannedPeripheral.peripheral
        return scannedPeripheral.peripheral
    }
}
```
Now I need to find the nearest peripheral according to these peripherals' average RSSI values of several seconds. I have no idea how to that. Could you give me some advice?
Thanks in advance.

paweljaneczek, 2018-01-15:
I am not quite sure if I understand your question. If you want to buffer all peripheral and then get the one with highest rssi consider using [buffer](http://reactivex.io/documentation/operators/buffer.html).

Anccerson, 2018-01-16:
@paweljaneczek Thanks! Problem solved. I use `.buffer(timeSpan: 5, count: 500, scheduler: scheduler).take(1)` to buffer all peripherals.

paweljaneczek, 2018-01-16:
Great to know!

closed 2018-01-16 by paweljaneczek