Polidea/RxBluetoothKit #199: Internal method `monitorPeripheral(on:peripheral:)` makes no sense.

DevAndArtist, 2017-12-26:
I'm currently digging through the source code to fully understand it's behavior especially the undocumented errors that can be thrown on whatever observables and I found this very frustrating issue:

```swift
func monitorPeripheral(
  on peripheralAction: Observable<RxPeripheralType>, 
  peripheral: Peripheral
) -> Observable<Peripheral> {
  let observable = peripheralAction
    .filter { $0 == peripheral.peripheral }
    .map { _ in peripheral }
  return ensure(.poweredOn, observable: observable)
}
```

This internal method is only used for convenience but the most annoying part is the last line `ensure(.poweredOn, observable: observable)` because it can cause the observable to be disposed with an error which shouldn't be there.

In theory we don't care what the bluetooth state is when observing the `connected/disconnected` state of the peripheral. That said `rx_isConnected` currently can error out, but shouldn't.

Furthermore the only other place where `monitorDisconnection(for:)` is used is already protected another call to `ensure(.poweredOn, observable: observable)` which means we can safely change the implementation above to:

```swift
func monitorPeripheral(
  on peripheralAction: Observable<RxPeripheralType>, 
  peripheral: Peripheral
) -> Observable<Peripheral> {
  return peripheralAction
    .filter { $0 == peripheral.peripheral }
    .map { _ in peripheral }
}
```

Cierpliwy, 2018-01-03:
When this code was tested on iOS, CBCentralManager didn't emit disconnected event for peripheral when Bluetooth was disabled on iPhone. However in such case we are sure that connection is lost and any observable which relies on connection should be completed with an error.

dariuszseweryn, 2018-02-24:
Closing due to inactivity

closed 2018-02-24 by dariuszseweryn