Polidea/RxBluetoothKit #204: Auto Pair

inovatek, 2018-01-16:
Hello

Can this library autopair programmatically ?

Thank You Very Much !!

paweljaneczek, 2018-01-17:
`RxBluetoothKit` is just an reactive wrapper around CoreBluetooth classes, so there is no autopair. If you want auto connect to device you could:
- connect to device and save it's uuid
- on every disconnect or app restart use this uuid to connect to that save device
Is this an answer for your question?

yakovzvi, 2018-01-17:
Is it possible to communicate to the ble device even without pairing ? Similar to a advertising tag ?

Thank You Very Much !!
Yanky

paweljaneczek, 2018-01-18:
Yes, check this answer https://stackoverflow.com/a/25255450

paweljaneczek, 2018-01-19:
For now closing. @yakovzvi If you have any other doubts, please reopen issue

closed 2018-01-19 by paweljaneczek