Polidea/RxBluetoothKit #205: (RxBluetoothKit.BluetoothError error 15.)

yusufonderd, 2018-01-17:
Hi all.
I'm getting RxBluetoothKit.BluetoothError when disable device bluetooth.
```

scanningDisposable = manager.rx_state
            .filter {  $0 == .poweredOn }
            .take(1)
            .timeout(3.0, scheduler:scheduler)
            .flatMap { _ in   
              return  self.manager.scanForPeripherals(withServices: [self.cbuuid], options:options) }
            .subscribeOn(MainScheduler.instance)
            .filter({ $0.peripheral.name != nil})
            .subscribe(onNext: { device in
                let peripheral  = device.peripheral
                self.connectPeripheral(peripheral: peripheral)
            }, onError: { error in
                print("scan error:\(error)")
            })
```
when I disable bluetooth,
logs:
scan error (RxBluetoothKit.BluetoothError error 15.)

paweljaneczek, 2018-01-19:
@yusufonderd I am not sure what is a problem here. It is correct that scanning returns `Bluetooth powered off` error when you disable device bluetooth. Am I missing something?

yusufonderd, 2018-01-19:
You can right but when I re-open bluetooth again. Code doesn't work.

paweljaneczek, 2018-01-19:
To be clear:
- you are starting scanning
- disabling bluetooth
- gets error
- enabling bluetooth

Is this what you doing? If yes, are you doing scanning again when bluetooth enabled?

paweljaneczek, 2018-01-19:
To be clear, this is not library related problem. The simplest solution would be to do the same (scanning) in `onError`

yusufonderd, 2018-01-19:
I know simplest solution. I think this code should continue scanning when toggle device bluetooth.

paweljaneczek, 2018-01-19:
Now scanning works in a way that when it receives change of state to other than `poweredOn` the scanning fails. Library is mainly a pure rx wrapper around CoreBluetooth methods without any additions. For now we are not planning to change this behaviour.

But it could be a nice addition, we will think about it in future, thanks!

closed 2018-01-24 by paweljaneczek