Polidea/RxBluetoothKit #224: Have you removed `listenOnRestoredState` completely?

DevAndArtist, 2018-03-14:
The new major release just broke my neck, I'm currently trying to update my app to support it. I used `listenOnRestoredState` to silent the annoying CoreBluetooth logs, but now this API is gone. My app isn't running yet so I cannot tell for sure if the warning will re-appear.

DevAndArtist, 2018-03-15:
Okay so it seems that the work is either not done yet:

- https://github.com/Polidea/RxBluetoothKit/blob/master/Source/CentralManager%2BRestoredState.swift#L9

- Or the documentation form the readme is old and entirely misleading:

> By giving proper identifier to CentralManager in constructor(options property), you can achieve state restoration functionality. Later, just make sure to *subscribe to listenOnRestoredState observable*, and inspect RestoredState instance, which consists any useful info about restored state.

Drusy, 2018-03-19:
Hey @minixT, any idea about the roadmap for a 5.0.x fixing this ?

minixT, 2018-03-20:
@Drusy we will fix this in the next version, which we plan to release soon.

Drusy, 2018-03-20:
@minixT Thanks for this, I have a related question, previously `listenOnRestoredState` was on Observable on which we could flatMap an `establishConnection` so a Disposable was returned and we could `dispose` it properly to close the connection. 

Since `RestoredState` can give us already connected peripherals, how can we get an `Observable<Peripheral>` to cancel those connections ?

Related to https://github.com/Polidea/RxBluetoothKit/issues/228#issuecomment-374555936

RamblinWreck77, 2018-03-20:
@Drusy  I'm having a similar problem from retrieveConnectedPeripherals

minixT, 2018-03-21:
@RamblinWreck77 Can you describe your problem in detail?

Drusy, 2018-03-22:
@minixT In the previous version it was possible to call `connect` on an already connected peripheral without error, so we could have an observable on every single `Peripheral`, wherever it comes (scan / restoration / retrieving)

Now the (re)connect an already connected peripheral is forbidden and no observable is returned from restoration as it will be a public constructor using block. Same problem for the two retrieving methods. So basically there is no way to disconnect those peripherals using RxBluetoothKit, the only way I see is to use the CBCentralManager directly `manager.centralManager.cancelPeripheralConnection(connectedPeripheral.peripheral)`

minixT, 2018-03-22:
@Drusy We will discuss all your doubts with the team and try to find the best solution.

minixT, 2018-03-23:
@Drusy Can you create a new issue with your questions and suggestions?

Drusy, 2018-03-23:
@minixT Done here https://github.com/Polidea/RxBluetoothKit/issues/238, hope it's clear enough ! I also pointed this subject discussed with pawelByszewski on Gitter

minixT, 2018-03-23:
Thanks @Drusy!

minixT, 2018-04-06:
Fixed in #237

closed 2018-04-06 by minixT