Polidea/RxBluetoothKit #228: 5.0 .cancelConnection() equivalent?

RamblinWreck77, 2018-03-14:
Hello! Congrats on the release!

I'm in the process of updating to the new API changes now and can't seem to find a replacement for .cancelConnection() on a peripheral.

Basically this class gets handed a peripheral, calls cancel connection, subscribes, then onNext notifies the app that the disconnect occurred. 

What's the 5.0 way to do this?

Thanks!

DevAndArtist, 2018-03-15:
I'm currently reading the code of 5.0 and from what I could gather so far the new connection is handled differently:

For a specific peripheral you can have at most one subscription for the connection. Then you have to maintain the disposable and cache it as long as you want you're peripheral to remain connected. (Or resubscribe on errors etc.)

If you want to cancel the connection you have simply to dispose that cached disposable from `establishConnection`.

https://github.com/Polidea/RxBluetoothKit/blob/master/Source/Connector.swift#L92

This actually makes it easier to work with because previously you had to maintain two different disposables if you wanted a clean `connect` / `disconnect` logic in your app.

minixT, 2018-03-15:
@DevAndArtist you are right. We've changed API and now peripheral is disconnecting automatically by the library, when you dispose disposable that is returned from the `establishConnection` method. There is no separate method to disconnect from peripheral.

DevAndArtist, 2018-03-15:
@minixT One thing I still do not understand. Can you elaborate on the logic of this `guard`?

https://github.com/Polidea/RxBluetoothKit/blob/master/Source/Connector.swift#L73

If I recall correctly previously you would just emit the same peripheral if the peripheral was already connected, but now it is an error?

RamblinWreck77, 2018-03-15:
@DevAndArtist this is causing a lot of problems for me as well. 

Personally I would prefer that code just re-emit the peripheral. I hope they change it!

DevAndArtist, 2018-03-15:
@RamblinWreck77 I think they can't because as I said before it's the users responsibility to maintain a single disposable for that. Just for such cases I build a [custom type](https://gist.github.com/DevAndArtist/7c0e885cb23c59414351d9a82b15d9c1) that hold's the disposable as long as it required and disposes it when it's replaced.

Just create an enum with a `connection` case and use `.disposed(by: disposeCache, key: .connection)` every time you call `establishConnection`. It should first dispose the old disposable and then try to re-connect.

DevAndArtist, 2018-03-19:
@minixT that also means that there won't be any errors thrown during disconnections like it was in 4.x right?

minixT, 2018-03-19:
Which exactly error do you mean?

DevAndArtist, 2018-03-19:
From my understanding of the internal code I came to the conclusion that a disconnection in previous version could potentially throw quite a few errors. I have a custom wrapper around the the `Peripheral` to allow easy error handling:

```swift
  /**
   * Completable can throw or complete:
   *  - .bluetoothNotPoweredOn
   *  - .disconnected
   *  - .notificationFailed
   */
  func disconnect<TriggerObservable>(
    handleError: ErrorHandler<TriggerObservable>? = nil,
    recover: RecoveryHandler? = nil,
    complete: CompletionHandler? = nil
  ) where TriggerObservable : ObservableType { ... }
```
Those are errors were simply mapped from `BluetoothError` for other reasons. The list is missing the `.destroyed` error because in my implementation it cannot happen.

minixT, 2018-03-20:
You can use `observeDisconnect` method in `CentralManager` class to get errors that happen during disconnection.

Drusy, 2018-03-20:
What about peripherals that have been retrieved using `retrieveConnectedPeripherals`, no `establishConnection` can be called as it will result in a error callback. How can we get a disposable out of this already-connected peripherals, did I miss something ?

DevAndArtist, 2018-03-20:
@minixT so there are possible errors during a disconnection attempt? Can you guys please document all the possible errors that you throw on the observables. I don't think the client should need do read the whole implementation to find out that critical information.

closed 2018-04-06 by minixT