Polidea/RxBluetoothKit #234: 2nd didUpdateValueFor never shows up

RamblinWreck77, 2018-03-20:
Hello! I'm running into a weird problem where the 2nd receive is timing out/never responding and I haven't been able to figure out why yet. Any help would be greatly appreciated! 

I've turned on the RxBluetoothKit debug log and thrown in some print statements.

The first step goes write > update notificationstate > didUpdateValueFor a-OK.  

``
RxBTSensorService.swift: SENSOR SEND: AT-Reset\r

2018-03-20 19:44:19.095: RxBTSensorService.swift:227 (connectToSensor(peripheral:)) -> Event next(Optional(RxBluetoothKit.Peripheral))

[RxBLEKit|DEBG|23:44:19.171]: Peripheral(uuid: AD2E6B66-99CD-F06B-07C4-0CCDE66A35E2, name: Optional("A")) didWriteValueFor(for:Characteristic(uuid: FFF2, id: 4728871376),
value: nil,
error: nil)

2018-03-20 19:44:19.181: RxBTSensorService.swift:238 (readFromConnectedSensor(terminateCharacter:)) -> subscribed

[RxBLEKit|DEBG|23:44:19.231]: Peripheral(uuid: AD2E6B66-99CD-F06B-07C4-0CCDE66A35E2, name: Optional("A")) didUpdateNotificationStateFor(
for:Characteristic(uuid: FFF1, id: 4728491392), isNotifying: true,
error: nil)

[RxBLEKit|DEBG|23:44:19.985]: Peripheral(uuid: AD2E6B66-99CD-F06B-07C4-0CCDE66A35E2, name: Optional("A")) didUpdateValueFor(for:Characteristic(uuid: FFF1, id: 4728491392),
value: Optional("0d0d454c4d3332372076312e350d0d3e"),
error: nil)

2018-03-20 19:44:19.990: RxBTSensorService.swift:238 (readFromConnectedSensor(terminateCharacter:)) -> Event next(RxBluetoothKit.Characteristic)

Returned: 16 bytes

2018-03-20 19:44:19.993: RxBTSensorService.swift:238 (readFromConnectedSensor(terminateCharacter:)) -> isDisposed

RxBTSensorService.swift: SENSOR RECV: \r\rRESET OK\r\r>

Write: 5 bytes

RxBTSensorService.swift: SENSOR SEND: AT-Next\r

[RxBLEKit|DEBG|23:44:20.074]: Peripheral(uuid: AD2E6B66-99CD-F06B-07C4-0CCDE66A35E2, name: Optional("A")) didUpdateNotificationStateFor(
for:Characteristic(uuid: FFF1, id: 4728491392), isNotifying: false,
error: nil)

[RxBLEKit|DEBG|23:44:20.134]: Peripheral(uuid: AD2E6B66-99CD-F06B-07C4-0CCDE66A35E2, name: Optional("A")) didWriteValueFor(for:Characteristic(uuid: FFF2, id: 4728871376),
value: nil,
error: nil)

2018-03-20 19:44:20.142: RxBTSensorService.swift:238 (readFromConnectedSensor(terminateCharacter:)) -> subscribed

[RxBLEKit|DEBG|23:44:20.193]: Peripheral(uuid: AD2E6B66-99CD-F06B-07C4-0CCDE66A35E2, name: Optional("A")) didUpdateNotificationStateFor(
for:Characteristic(uuid: FFF1, id: 4728491392), isNotifying: true,
error: nil)
``

Read/Write are FF1/FF2, and Write shows up as Notify + Indicate on LightBlue. Am I making some newbie mistake here or is this a bug?

Thanks!

RamblinWreck77, 2018-03-21:
Ok now this is super weird!

This behavior occurs on our new hardware, and not on our old hardware. 

Both have a service FFF0, and read (notify indicate) FFF1 and write FFF2

The only difference is the hardware string, BT5050 (works) vs BT5056 (exhibits the above behavior). 

Both Devices can be communicated with on 4.x versions of RxBluetoothkit

minixT, 2018-03-21:
Hi @RamblinWreck77!

Can you put here a code how you register on characteristic notifications?

RamblinWreck77, 2018-03-21:
@minixT I got to the bottom of it, the issue was the new write code was not setting notify on read *before* writing. 

Write -> observeValueUpdateAndSetNotification produced odd behavior that worked on some devices and not others

Manually accessing the Peripheral's CBPeripheral and setting .setNotifyValue(true, for: readCharacteristic), THEN writing, THEN running .observeValueUpdate seemed to solve it for all of our devices.

closed 2018-03-21 by RamblinWreck77