Polidea/RxBluetoothKit #235: [5.0] Reading and observing values using CharacteristicIdentifier onError before didDiscoverCharacteristics

Drusy, 2018-03-22:
Since v5.0, chaining `establishConnection` and `observeValueUpdateAndSetNotification`/`readValue` produces the RxSwift error `Sequence doesn't contain any elements`.

### Conditions

- Happens when chaining instantly the read or/and notify for more than once characteristic.
- Always succeed after few retry (1~3 x2s)
- Fails only after `establishConnection`, always works if the peripheral comes from `retrieveConnectedPeripherals` (services & characteristics already in memory ?)

### Related code:

```
enum DeviceCharacteristic: String, CharacteristicIdentifier {
    
    // Monitoring
    case temperature = "1b0d1202-a720-f7e9-46b6-31b601c4fca1" // °C
    case battery = "1b0d1203-a720-f7e9-46b6-31b601c4fca1" // mV

    var uuid: CBUUID {
        return CBUUID(string: self.rawValue)
    }
    
    //Service to which characteristic belongs
    var service: ServiceIdentifier {
        switch self {
        case .temperature, .battery:
            return DeviceService.monitoring
    }
}

enum DeviceService: String, ServiceIdentifier {
    case monitoring = "1b0d1200-a720-f7e9-46b6-31b601c4fca1"

    var uuid: CBUUID {
        return CBUUID(string: self.rawValue)
    }
}
```

```
self.manager.observeState()
    .startWith(manager.state)
    .filter { $0 == .poweredOn }
    .take(1)
    .flatMap {  peripheral.establishConnection() }
    .subscribe(
        onNext: { connectedPeripheral in
            self.onDeviceConnected(for: connectedPeripheral)
    },
        onError: { error in
            print("❌ Could not connect \(peripheral.name ?? ""): \(error)")
    }).disposed(by: self.currentConnectionDisposeBag)
```

``` 
func onDeviceConnected(for: Peripheral) {
    print("🔵 \(peripheral.name ?? "") is now connected")

    // Read
    PrimitiveSequence.zip(
        peripheral.readValue(for: DeviceCharacteristic.battery).do(onSuccess: { _ in
            print("✅ Read battery")
        }),
        peripheral.readValue(for: DeviceCharacteristic.temperature).do(onSuccess: { _ in
            print("✅ Read temperature")
        }))
        .retryWhen { error in
            error.do(onNext: { error in
                print("❌ An error occured reading the monitoring service information: \(error) (will retry in 2s)")
            }).delay(2, scheduler: MainScheduler.instance)
        }
        .subscribeOn(MainScheduler.instance)
        .subscribe()
         disposed(by: currentConnectionDisposeBag)

    // Notify
    Observable.combineLatest([
        peripheral.observeValueUpdateAndSetNotification(for: DeviceCharacteristic.battery).do(onNext: { _ in
            print("✅ Updated battery")
        }),
        peripheral.observeValueUpdateAndSetNotification(for: DeviceCharacteristic.temperature).do(onNext: { _ in
            print("✅ Updated temperature")
        })])
        .retryWhen { error in
            error.do(onNext: { error in
                print("❌ An error occured subscribing to notification for the monitoring service: \(error) (will retry in 2s)")
            }).delay(2, scheduler: MainScheduler.instance)
        }
        .subscribeOn(MainScheduler.instance)
        .subscribe()
        .disposed(by: currentConnectionDisposeBag)
```

### Produced logs:

```
🔵 DW3_12 is now connected
[RxBLEKit|DEBG|09:14:58.501]: Peripheral(uuid: EA1C79F7-7484-06FA-10BC-6A2F1B0C476B, name: Optional("DW3_12")) didDiscoverServices(services
: Optional("[Service(uuid: 1B0D1200-A720-F7E9-46B6-31B601C4FCA1, id: 4461714176)]"),
error: nil)
[RxBLEKit|DEBG|09:14:58.514]: Peripheral(uuid: EA1C79F7-7484-06FA-10BC-6A2F1B0C476B, name: Optional("DW3_12")) didDiscoverServices(services
: Optional("[Service(uuid: 1B0D1200-A720-F7E9-46B6-31B601C4FCA1, id: 4461714176)]"),
error: nil)
[RxBLEKit|DEBG|09:14:58.515]: Peripheral(uuid: EA1C79F7-7484-06FA-10BC-6A2F1B0C476B, name: Optional("DW3_12")) didDiscoverServices(services
: Optional("[Service(uuid: 1B0D1200-A720-F7E9-46B6-31B601C4FCA1, id: 4461714176)]"),
error: nil)
[RxBLEKit|DEBG|09:14:58.515]: Peripheral(uuid: EA1C79F7-7484-06FA-10BC-6A2F1B0C476B, name: Optional("DW3_12")) didDiscoverServices(services
: Optional("[Service(uuid: 1B0D1200-A720-F7E9-46B6-31B601C4FCA1, id: 4461714176)]"),
error: nil)
[RxBLEKit|DEBG|09:14:58.981]: Peripheral(uuid: EA1C79F7-7484-06FA-10BC-6A2F1B0C476B, name: Optional("DW3_12")) didDiscoverCharacteristicsFor(for:
Service(uuid: 1B0D1200-A720-F7E9-46B6-31B601C4FCA1, id: 4461714176), characteristics:
Optional("[Characteristic(uuid: 1B0D1203-A720-F7E9-46B6-31B601C4FCA1, id: 4442392224)]"),
error: nil)
❌ An error occured reading the monitoring service information: Sequence doesn't contain any elements. (will retry in 2s)
❌ An error occured subscribing to notification for the monitoring service: Sequence doesn't contain any elements. (will retry in 2s)
[RxBLEKit|DEBG|09:14:58.996]: Peripheral(uuid: EA1C79F7-7484-06FA-10BC-6A2F1B0C476B, name: Optional("DW3_12")) didDiscoverCharacteristicsFor(for:
Service(uuid: 1B0D1200-A720-F7E9-46B6-31B601C4FCA1, id: 4461714176), characteristics:
Optional("[Characteristic(uuid: 1B0D1203-A720-F7E9-46B6-31B601C4FCA1, id: 4442392224), Characteristic(uuid: 1B0D1202-A720-F7E9-46B6-31B601C4FCA1, id: 4461906064)]"),
error: nil)
[RxBLEKit|DEBG|09:14:58.998]: Peripheral(uuid: EA1C79F7-7484-06FA-10BC-6A2F1B0C476B, name: Optional("DW3_12")) didDiscoverCharacteristicsFor(for:
Service(uuid: 1B0D1200-A720-F7E9-46B6-31B601C4FCA1, id: 4461714176), characteristics:
Optional("[Characteristic(uuid: 1B0D1203-A720-F7E9-46B6-31B601C4FCA1, id: 4442392224), Characteristic(uuid: 1B0D1202-A720-F7E9-46B6-31B601C4FCA1, id: 4461906064)]"),
error: nil)
[RxBLEKit|DEBG|09:14:58.998]: Peripheral(uuid: EA1C79F7-7484-06FA-10BC-6A2F1B0C476B, name: Optional("DW3_12")) didDiscoverCharacteristicsFor(for:
Service(uuid: 1B0D1200-A720-F7E9-46B6-31B601C4FCA1, id: 4461714176), characteristics:
Optional("[Characteristic(uuid: 1B0D1203-A720-F7E9-46B6-31B601C4FCA1, id: 4442392224), Characteristic(uuid: 1B0D1202-A720-F7E9-46B6-31B601C4FCA1, id: 4461906064)]"),
error: nil)
[RxBLEKit|DEBG|09:14:59.039]: Peripheral(uuid: EA1C79F7-7484-06FA-10BC-6A2F1B0C476B, name: Optional("DW3_12")) didUpdateValueFor(for:Characteristic(uuid: 1B0D1203-A720-F7E9-46B6-31B601C4FCA1, id: 4442392224),
value: Optional("f609"),
error: nil)
[RxBLEKit|DEBG|09:14:59.159]: Peripheral(uuid: EA1C79F7-7484-06FA-10BC-6A2F1B0C476B, name: Optional("DW3_12")) didUpdateNotificationStateFor(
for:Characteristic(uuid: 1B0D1203-A720-F7E9-46B6-31B601C4FCA1, id: 4442392224), isNotifying: true,
error: nil)
[RxBLEKit|DEBG|09:14:59.219]: Peripheral(uuid: EA1C79F7-7484-06FA-10BC-6A2F1B0C476B, name: Optional("DW3_12")) didUpdateNotificationStateFor(
for:Characteristic(uuid: 1B0D1203-A720-F7E9-46B6-31B601C4FCA1, id: 4442392224), isNotifying: false,
error: nil)
[RxBLEKit|DEBG|09:15:01.049]: Peripheral(uuid: EA1C79F7-7484-06FA-10BC-6A2F1B0C476B, name: Optional("DW3_12")) didUpdateValueFor(for:Characteristic(uuid: 1B0D1203-A720-F7E9-46B6-31B601C4FCA1, id: 4442392224),
value: Optional("f609"),
error: nil)
✅ Read battery
✅ Updated battery
[RxBLEKit|DEBG|09:15:01.109]: Peripheral(uuid: EA1C79F7-7484-06FA-10BC-6A2F1B0C476B, name: Optional("DW3_12")) didUpdateValueFor(for:Characteristic(uuid: 1B0D1202-A720-F7E9-46B6-31B601C4FCA1, id: 4461906064),
value: Optional("14"),
error: nil)
✅ Read temperature
✅ Updated temperature
```

Drusy, 2018-03-22:
### Complementary hints

- Same code was working with v4
- Not related to `zip` / `combineLatest` as I can reproduce the same errors simply calling more than one read/notify one by one

Drusy, 2018-04-03:
Is it the intended behaviour ? I mean it's obvious that you want to wait until the services / characteristics are (at least have been tried to be)  retrieved. In case there is an error during retrieving a service / characteristic, I would understand, but it's not the case.

For me it's a main breaking change and I would loose interest using `CharacteristicIdentifier` and `ServiceIdentifier` which for me is the first reason I love RxBluetoothKit :)

DevAndArtist, 2018-04-03:
Well I cannot answer that but I fixed that issue in my app right now by just retry after 2 seconds if this issue happens and wait until the peripheral is connected before it will retry.

(old) https://github.com/Polidea/RxBluetoothKit/blob/eb5294486e7ca37ae3e36e13375efbb8e55bba45/Source/Peripheral.swift#L204

(new) https://github.com/Polidea/RxBluetoothKit/blob/master/Source/Peripheral.swift#L215

minixT, 2018-04-04:
This situation should not happen. This will be fixed.

minixT, 2018-04-09:
Fixed in #244

closed 2018-04-09 by minixT