Polidea/RxBluetoothKit #236: [5.0] Regression behind the redesigned `scanForPeripherals`

DevAndArtist, 2018-03-22:
The new `scanForPeripherals` method comes with a huge regression. According to the readme scans were queued and shared before. I understand the idea that per manager there should be only one active scan, but the issue is that if you need to access `ScannedPeripheral`'s from different places you end up in a dead end.

For instance we have a library that might want to scan for a short period of time, but also our app that imports that library needs to able to scan independently. Before 5.0 we had a centralized way to access the `ScannedPeripheral`'s but now we don't.

To work around the issue I had to implement a stored property on `CentralManager` extension using Objective-C runtime because the class itself is `public`, which is potentially dangerous. This particular stored property is of type `Scanner` which routes the scan into one `PublishSubject` which can have more then one observer. The scan is stopped in a similar fashion like the new value observation logic on a characteristic, so to speak *when all previous observer are disposed*. 

Since it's too late to add a huge breaking change for 5.0 I suggest a different solution that would at least allow us to implement a safe workaround.

Make `CentralManager` `open` and allow custom subclasses.

DevAndArtist, 2018-04-04:
Here is what I came up with to workaround, but I wish I code move the property directly into a custom subclass of `CentralManager` or the official design to be similar to this.

```swift
extension CentralManager {
  ///
  private static var centralManagerScannerExtension
    = "centralManagerScannerExtension"

  ///
  public var scanner: Scanner {
    get {
      if let scanner = objc_getAssociatedObject(
          self,
          &CentralManager.centralManagerScannerExtension
        ) as? Scanner {
        return scanner
      } else {
        let newScanner = Scanner(manager: self)
        self.scanner = newScanner
        return newScanner
      }
    }
    set {
      objc_setAssociatedObject(
        self,
        &CentralManager.centralManagerScannerExtension,
        newValue,
        objc_AssociationPolicy.OBJC_ASSOCIATION_RETAIN_NONATOMIC
      )
    }
  }
}
```

```swift
public final class Scanner {
  ///
  private var disposeBag = DisposeBag()

  ///
  private let relay = PublishRelay<ScannedPeripheral>()

  ///
  private let lock = NSRecursiveLock()

  ///
  private var activeObservable: Observable<Scanner>?

  ///
  private var isScanning = false

  ///
  private var activeSubscriptions = 0

  ///
  public unowned let manager: CentralManager

  ///
  public enum Error : Swift.Error {
    case destroyed
  }

  ///
  public init(manager: CentralManager) {
    self.manager = manager
  }

  ///
  public func scanForPeripherals(
    withServices serviceUUIDs: [CBUUID]?,
    options: [String: Any]?
  ) -> Observable<Scanner> {
    // Create a new observable or extract the existing one
    return .deferred { [weak self] in
      // Unwrap self if possible
      guard let scanner = self else {
        throw Scanner.Error.destroyed
      }
      // Guard the thread
      scanner.lock.lock()
      defer { scanner.lock.unlock() }
      // Return an existing shared observable
      if let activeObservable = scanner.activeObservable {
        return activeObservable
      }
      // Create a new observable
      let observable = Observable<Scanner>
        .create { [unowned scanner] in
          $0.onNext(scanner)
          return Disposables.create()
        }
        .do(
          onSubscribed: { [weak self] in
            guard let scanner = self else { return }
            scanner.lock.lock()
            defer { scanner.lock.unlock() }
            // If not yet started scaning then subscribe
            if scanner.isScanning == false {
              // Set the flag
              scanner.isScanning = true
              // Create a new scan with new parameters
              scanner.manager
                .scanForPeripherals(
                  withServices: serviceUUIDs,
                  options: options
                )
                .subscribe(
                  onNext: { [unowned scanner] in
                    scanner.relay.accept($0)
                  },
                  onError: { [unowned scanner] _ in
                    scanner.disposeBag = DisposeBag()
                    scanner.isScanning = false
                  }
                )
                .disposed(by: scanner.disposeBag)
            }
            // Increment the count of subscriptions
            scanner.activeSubscriptions += 1
          },
          onDispose: { [weak self] in
            guard let scanner = self else { return }
            scanner.lock.lock()
            defer { scanner.lock.unlock() }
            // Decriment the count of subscriptions
            scanner.activeSubscriptions -= 1
            // Stop the scan if the count wen to 0
            if scanner.activeSubscriptions <= 0 {
              scanner.disposeBag = DisposeBag()
              scanner.isScanning = false
              scanner.activeObservable = nil
            }
        })
        .share()
      scanner.activeObservable = observable
      return observable
    }
  }

  ///
  public func observeScannedPeripherals() -> Observable<ScannedPeripheral> {
    return relay.asObservable()
  }
}
```

paweljaneczek, 2018-04-06:
We think that current approach for `scanForPeripherals` is a correct one and for now we are not going to change it (and from what you wrote that's fine for you).

We would also rather not like to make `CentralManager` open. It sounds like a hack to fix issue that is not related to library.

What we would like to propose you is to create your own fork and make `CentralManager` open. This way you could handle this in a clean way.

closed 2018-06-20 by paweljaneczek