Polidea/RxBluetoothKit #238: [5.0] Unable to get an Observable<Peripheral> from already connected peripheral provided by the system

Drusy, 2018-03-23:
In v4 it was possible to call `connect` on an already connected peripheral without error, so we could have an observable on every single Peripheral, wherever it comes (scan / restoration / retrieving)

Now calling `establishConnection` on an already connected peripheral is forbidden. So there is no way to get an `Observable<Peripheral>` from peripherals that comes from state restoration as it will provider a `[Peripheral]`.

Same problem for `retrieveConnectedPeripherals(_) -> [Peripheral]`. So basically there is no way to disconnect those peripherals using RxBluetoothKit, the only way I see is to use the CBCentralManager directly and call `manager.centralManager.cancelPeripheralConnection(connectedPeripheral.peripheral)`

Discussed here https://gitter.im/RxBLELibraries/RxBluetoothKit with @pawelByszewski:

> We are thinking of making possible to call establishConnection on already connected peripheral. It would work the same as reading characteristic value, so we would do ref counting on subscriptions and when ref count goes to 0 then dispose & disconnect happens.
> Those solutions would not prevent user from their mistakes (need to dispose every subscription to disconnect device), but would surely give them more freedom and would fix problems that we currently have.
> What you think about it? We really want to make library that is easy to use and gives all the possibility as CBCentralManager gives.

> We will need to discuss that problem internally at Polidea and will give some final proposition next week

RamblinWreck77, 2018-03-25:
In my project that's exactly what I had to do, drill down to central manager and disconnect from the peripheral if I happened to already be connected.  Not a deal breaker, but definitely not ideal... and it kind of defeats the purpose of 5.0's auto-disconnect feature. Since this was necessary in 4.x anyways I was already used to doing this. 

I'm interested to see how they evolve this over time.

DevAndArtist, 2018-03-29:
@minixT I wonder if we can fix the `establishConnection` function by doing something like this (not tested code):

```swift
fileprivate func createConnectionObservable(
    for peripheral: Peripheral,
    options: [String: Any]? = nil
) -> Observable<Peripheral> {
  return Observable.create { [weak self] observer in
    guard let strongSelf = self else {
      observer.onError(BluetoothError.destroyed)
      return Disposables.create()
    }

    let connectingStarted = strongSelf.connectingBox.compareAndSet(
      compare: { !$0.contains(peripheral.identifier) },
      set: { $0.insert(peripheral.identifier) }
    )

    guard connectingStarted else {
      observer.onError(
        BluetoothError.peripheralIsConnectingOrAlreadyConnected(peripheral)
      )
      return Disposables.create()
    }

    let connectedObservable = strongSelf
      .createConnectedObservable(for: peripheral)
    let failToConnectObservable = strongSelf
      .createFailToConnectObservable(for: peripheral)
    let disconnectedObservable = strongSelf
      .createDisconnectedObservable(for: peripheral)

    var observable = connectedObservable.amb(failToConnectObservable)
      .do(onNext: { observer.onNext($0) })
      .flatMap { _ in disconnectedObservable }

    if peripheral.isConnected {
      observable = observable.startWith(peripheral)
    }

    let disposable = observable
      .subscribe(onError: { observer.onError($0) })

    if !peripheral.isConnected {
      strongSelf.centralManager.connect(peripheral.peripheral, options: options)
    }

    return Disposables.create { [weak self] in
      guard let strongSelf = self else { return }
      disposable.dispose()
      let isConnecting = strongSelf.connectingBox.read { 
        $0.contains(peripheral.identifier) 
      }
      if isConnecting || peripheral.isConnected {
        strongSelf.disconnectingBox.write { $0.insert(peripheral.identifier) }
        strongSelf.centralManager.cancelPeripheralConnection(peripheral.peripheral)
        strongSelf.connectingBox.write { $0.remove(peripheral.identifier) }
      }
    }
  }
}
```

minixT, 2018-04-03:
@DevAndArtist We discussed current situation in the team and we want to add a change similar to what you have proposed. We will return an error only if app was connected to the peripheral using the library, what will enable disconnecting when peripheral comes from state restoration. It will still be impossible to call establishConnection second time for one peripheral when previous observable was not disposed.

DevAndArtist, 2018-04-03:
@minixT keep in mind it was only some bikeshedding which does not work because the amount of observables is not tracked.

commit 622bd2aa (2018-04-06): Enabled getting an Observable<Peripheral> from already connected peripheral provided by the system.

Fix #238
closed 2018-04-06 by minixT