Polidea/RxBluetoothKit #239: Write to characteristic does not work

aleksanderaleksic, 2018-03-25:
Hi! I am trying to write to a Characteristic, but it doesn't seem to work. 
If I write directly on the peripheral and passing the `CharacteristicIdentifier ` it works.

## **CODE**
This is the code i want to execute
```
func playBeepSound() -> Observable<Void> {
    let command = ATCoreBeepForPairing()
    let obs = self.characteristic
        .flatMap{ $0.observeValueUpdateAndSetNotification() }
        .filter{
            return command.successfullInstrumentCommand(data: $0.value)
        }
        .take(1)
        .map { (_) -> Void in
            return
        }
    
    self.characteristic
        .flatMap { $0.writeValue(command.rawData, type: .withResponse) }
        .subscribe()
        .disposed(by: self.bag)
    
    return obs
}
```

## **TEST CODE**
```
func test_playBeepSound(){
    let expectation = XCTestExpectation(description: "Play Beep Sound")
    
    func beep() {
        let command = ATCoreBeepForPairing()
        
        self.instrument.peripheral
            .flatMap { $0.writeValue(command.rawData, for: CoreCharacteristic.asrcap, type: .withResponse) }
            .subscribe()
            .disposed(by: self.bag)
    }
    
    let beepSound = self.instrument.asracp.playBeepSound()
    
    beepSound
        .subscribe(onNext: { (_) in
            expectation.fulfill()
        }, onError: { (error) in
            XCTFail("A error accured: \(error)")
        }).disposed(by: self.bag)

    beep()

    wait(for: [expectation], timeout: 20.0)
}
```
In the test code, i excecute the `self.instrument.asracp.playBeepSound()` witch is the **CODE** on top. It does not write anything to the characteristic. When run the `beep()` function, the device beeps and the test succeds. The `beep()` does exacly the same as the `self.instrument.asracp.playBeepSound()` except it writes on the peripheral.

aleksanderaleksic, 2018-03-25:
If i don't run `beep()` the observable doesn't emit anything. And the test times out.

aleksanderaleksic, 2018-04-02:
I think this is a mistake by me. Since it gets disposed immediately.
Haven't figured it out yet, any suggestions?

```
[AIRTHINGS.BLE]-[VERBOSE]-[2018-04-02 12:42:21 +0000] Manager state:  poweredOn 
[AIRTHINGS.BLE]-[VERBOSE]-[2018-04-02 12:42:21 +0000] Manager started scanning for peripherals 
[AIRTHINGS.BLE]-[VERBOSE]-[2018-04-02 12:42:22 +0000] Found instrument with serial:  2900123243 
[AIRTHINGS.BLE]-[VERBOSE]-[2018-04-02 12:42:22 +0000] Connected to:  2900123243 
[AIRTHINGS.BLE]-[VERBOSE]-[2018-04-02 12:42:22 +0000] -[SERVICE] On Subscribe 
[AIRTHINGS.BLE]-[VERBOSE]-[2018-04-02 12:42:22 +0000] -[SERVICE] On Complete 
[AIRTHINGS.BLE]-[VERBOSE]-[2018-04-02 12:42:22 +0000] -[SERVICE] On Subscribed 
[AIRTHINGS.BLE]-[VERBOSE]-[2018-04-02 12:42:22 +0000] -[SERVICE] On Dispose 
[AIRTHINGS.BLE]-[VERBOSE]-[2018-04-02 12:42:22 +0000] -[CHARS] On Subscribe 
[AIRTHINGS.BLE]-[VERBOSE]-[2018-04-02 12:42:22 +0000] -[CHARS] On Complete 
[AIRTHINGS.BLE]-[VERBOSE]-[2018-04-02 12:42:22 +0000] -[CHARS] On Subscribed 
[AIRTHINGS.BLE]-[VERBOSE]-[2018-04-02 12:42:22 +0000] -[CHARS] On Dispose 
[AIRTHINGS.BLE]-[VERBOSE]-[2018-04-02 12:42:22 +0000] -[ASRACP] On Subscribe 
[AIRTHINGS.BLE]-[VERBOSE]-[2018-04-02 12:42:22 +0000] -[ASRACP] On Complete 
[AIRTHINGS.BLE]-[VERBOSE]-[2018-04-02 12:42:22 +0000] -[ASRACP] On Subscribed 
[AIRTHINGS.BLE]-[VERBOSE]-[2018-04-02 12:42:22 +0000] -[ASRACP] On Dispose 
```

aleksanderaleksic, 2018-04-07:
@kkosovsky Thanks for your reply. Now the test actually succeeds, yeey! But, there should be a beep sound from the device, but this never happens.

aleksanderaleksic, 2018-04-08:
I made it work! The problem was that the writing happens before the characteristic was set up.
So iIset the characteristic subject as a `ReplaySubject` instead with the buffer size of 1.
`private let _asracpCharSubject = ReplaySubject<ASRACP>.create(bufferSize: 1)`

This plays the beep sound and the test succeeds. But is there any disadvantages doing it this way?

kkosovsky, 2018-04-10:
@aleksanderaleksic I'm glad you've found my suggestion helpful :) Can you paste a bigger code snippet, so I can actually see what's going on in detail? Then it would be easier, to advise you with the best type of subject for your problem :) Cheers!

paweljaneczek, 2018-04-20:
@aleksanderaleksic was the problem solved?

aleksanderaleksic, 2018-04-27:
Sorry for the late reply. 
@kkosovsky I don't have the time go into detail on what I did, but the problem was solved for me by using the `ReplaySubject`.

Thanks for the help.

closed 2018-04-27 by aleksanderaleksic