Polidea/RxBluetoothKit #246: Cocoapods Specification for RxBluetoothKit 4.0

bleeckerj, 2018-04-09:
Hello

We're not ready to migrate to RxBluetoothKit 5.0 — there are several breaking changes that would require an extensive refactoring.

When I attempt to specific RxBluetoothKit 4.0 in my Podfile, 5.0 is still downloaded. I wonder if you could help point out how my specification may be wrong?

`pod 'RxBluetoothKit', :git => 'https://github.com/Polidea/RxBluetoothKit.git', :branch => 'master', :tag => '4.0.2'`

Thank you for your help.

Julian

minixT, 2018-04-10:
Hi @bleeckerj, you can use this code in your `Podfile`:

```
pod 'RxBluetoothKit', '~>4.0'
```

It will always install the newest `4.x` version and will not update it to `5.x` od higher.

[Here](https://guides.cocoapods.org/using/the-podfile.html#migrating-from-x-to-) you can find more about pods versions.

paweljaneczek, 2018-04-18:
@bleeckerj did it solve your issue?

bleeckerj, 2018-04-18:
@paweljaneczek Yes, it did. Thank you. I'll Close.

closed 2018-04-18 by bleeckerj