Polidea/RxBluetoothKit #247: Speed problem investigation and inconsisent write .withoutResponse behavior

RamblinWreck77, 2018-04-10:
Hello! I'm currently working on a project that needs ~10hz R/W's over to a peripheral. I'm not getting that speed and that investigation has lead me down a rabbit hole. 

I've narrowed it down to one line of code causing most of the problems:

```
.writeValue(data, for: MyExampleCharacteristicIdentifier.write, type: .withResponse)
```

Takes ~ 0.15-0.215 seconds. Changing nothing else

```
.writeValue(data, for: MyExampleCharacteristicIdentifier.write, type: .withoutResponse)
```

takes ~ 0.003 seconds. It seems fairly straightforward that withoutResponse is faster, but is this level of speed difference normal? 

I then wanted to investigate if the new protocols/caching was responsible but the results were similar.

The main issue that prevents me from simply using .withoutResponse is I seem to be encountering inconsistent events with .withoutResponse. The docs say it should emit next immediately after writing, however in this case with .debug() on and a 5-second timeout nothing happens.

During the same session in the same code it searches, finds a sensor, writes, and times out with next never occuring. A few seconds later (the 2nd attempt) when connecting again next fires as expected and everything is fine. 

Here's my cleaned-up log of the event:

```
WRITE START: 1523319438.70003
SENSOR SEND: ATZ\r
2018-04-09 20:17:18.702: SensorService.swift:361 (writeToConnectedSensor(data:terminateCharacter:timeout:)) -> subscribed
...
(5s later)
...
SensorService.swift: 🔥 Error in writeValue: {SensorServiceError}: Write Timed Out
{SensorServiceError}: Disconnected
2018-04-09 20:17:28.712: SensorService.swift:361 (writeToConnectedSensor(data:terminateCharacter:timeout:)) -> isDisposed
...
... (about to try again)
...
WRITE START: 1523319457.25237
SENSOR SEND: ATZ\r
2018-04-09 20:17:37.254: SensorService.swift:361 (writeToConnectedSensor(data:terminateCharacter:timeout:)) -> subscribed
2018-04-09 20:17:37.258: SensorService.swift:361 (writeToConnectedSensor(data:terminateCharacter:timeout:)) -> Event next(
...
... And everything works!
...
```

Same session, same code, and I can't figure out why sometimes Event next( gets emitted and sometimes not. The only pattern I've seen is if it fails on the first try, it always succeeds immediately afterwards. 

I wish I could be more helpful here... if I can provide any more information to assist please just ask. Any help is greatly appreciated!

RamblinWreck77, 2018-04-10:
On further investigation this behavior only occurs when I call RxBluetooth Peripheral .writeValue()

A (terribly hacky):

```
let charW = service.characteristics!.first(where: {$0.uuid == MyExampleCharacteristicIdentifier.write.uuid})!
 let obj = Observable.of(peri.peripheral.writeValue(data, for: charW.characteristic, type: .withoutResponse))
return obj
```

Does not product the above behavior, I can write every time without fail and write speeds are in the ~0.0015 - 0.0020s range.

RamblinWreck77, 2018-04-10:
Sorry for the spam, but I'm seeing similar delays when reading as well:

```
WRITE END: 1523326407.29391
READ START: 1523326407.29398
MAP END: 0.17071795463562 later
MAP END: 0.231263875961304 later
```

for

```
return peri.observeValueUpdate(for: ExampleCharacteristic.read)
                                .timeout(timeout, other: Observable.error(SensorServiceError.readTimeout), scheduler: MainScheduler.instance)
                                .map { char in
                                    print("MAP END: \(Date().timeIntervalSince1970 - start) later")
                                    return char.value ?? Data()
                                }
```

paweljaneczek, 2018-04-18:
Hey @RamblinWreck77, sorry for such big delay.

In our library, before doing `writeValue` we are checking if `CBPeripheral.canSendWriteWithoutResponse` value is set to true. If it doesn't then we start observing for `CBPeripheralDelegate.peripheralIsReady(toSendWriteWithoutResponse peripheral: CBPeripheral)`.
Possibly that in our case that value is set to false and we never get delegate method being called (or it is being called after timeout).

Could you set logging with `RxBluetoothKitLog.setLogLevel(.verbose)` and print `CBPeripheral.canSendWriteWithoutResponse` before you are doing write call?
This could help us in resolving this issue.

closed 2018-04-18 by RamblinWreck77

davide-esr, 2019-04-11:
I ran into a very similar issue and I only solved it after making sure that I was using the same dispatch queue for the central manager and for subscribing to `writeValue`