Polidea/RxBluetoothKit #248: observeConnection

LiuSky, 2018-04-16:
I used to use 4 rx_isConnected for a good time. But with a 5 observeConnection, this will automatically call the disconnected method observeDisconnect after my active connection. I don't know what it was. I'm confused. @minixT

LiuSky, 2018-04-16:
I will automatically disconnect when I connect with an event, but the 4 version is normal. Do not take the initiative to disconnect. I don't know what logic has caused by 5. @minixT

LiuSky, 2018-04-16:
When the connection is successful, a callback method is returned within a few seconds. 4.0 does not.

LiuSky, 2018-04-16:
Thank you for your reply. I want to know why it is designed. gratitude @paweljaneczek

paweljaneczek, 2018-04-16:
There were several reasons of that:
- simplified use of api - now there is no need of calling separately `disconnect`, it is done automatically when disposing
- it is more reactive style of coding
- have the same api as our android library, which will be easier for us to maintain it and reuse it in our react-native and flutter libraries

LiuSky, 2018-04-16:
Is there a way to take the initiative? @paweljaneczek

paweljaneczek, 2018-04-16:
Not sure what you mean. Would you like to contribute to library?

LiuSky, 2018-04-16:
5.0 version, I just started reading in the afternoon, but I don't know the logic of 5.0 at the moment.

LiuSky, 2018-04-16:
Is there any way I can't take the initiative to disconnect when I call establishConnection? @paweljaneczek  Thank you

paweljaneczek, 2018-04-16:
The only way to disconnect (using our library) is to dispose `observeConnection` subscription.

LiuSky, 2018-04-16:
I'm sorry, maybe my expression is not very clear. My problem is that I actively invoke the establishConnection method, and after the link is successful, he will take the initiative to disconnect the connection, rather than I call it manually.
[image]
[image]

paweljaneczek, 2018-04-16:
could you also share code when you are subscribing for that observable?

LiuSky, 2018-04-16:
Sorry, I don't quite understand your meaning, or wait for the sample project. Thank you @paweljaneczek

closed 2018-04-16 by LiuSky