Polidea/RxBluetoothKit #267: Central manager unknown state

aleksanderaleksic, 2018-06-20:
I have an issue with that the state observable for the Central Manager don't emit any state.
I try to scan for peripherals, but it times out before it gets there. 

```
private let queue = DispatchQueue(label: "airthings.ble")
private let manager : CentralManager!
private var scanDisposable: Disposable?

private init() {
        self.manager = CentralManager(queue: queue, options: nil)
}

func scan(for types: [SupportedInstrumentTypes]) -> Observable<ATScannedInstrument> {
        let services = SupportedInstrumentTypes.allValues.flatMap { (type) -> CBUUID? in
            return type.service
        }
        
        if scanDisposable == nil {
            scanDisposable = manager.observeState()
                .startWith(manager.state)
                .internalLog(identifier: "MANAGER STATE")
                .filter{ $0 == .poweredOn }
                .timeout(3.0, scheduler: MainScheduler.instance)
                .onTimeout({ (error) -> Observable<BluetoothState> in
                    let err = BLEError(description: "Unable to read bluetooth state")
                    return .error(ATBleError.timeout(err))
                })
                .take(1)
                .flatMap({ (_) -> Observable<ScannedPeripheral> in
                    return self.manager.scanForPeripherals(withServices: services)
                })
                ...
                ...
                ...
}

This generates this log output:

```
[RxBLEKit|DEBG|08:54:16.960]: CentralManager(7791418752) didUpdateState(state: poweredOn)
[RxBLEKit|DEBG|08:54:16.961]: CentralManager(7793447552) didUpdateState(state: poweredOn)
[AIRTHINGS.BLE]-[VERBOSE]-[2018-06-20 08:54:16 +0000] -[MANAGER STATE] On Subscribe 
[AIRTHINGS.BLE]-[VERBOSE]-[2018-06-20 08:54:16 +0000] -[MANAGER STATE] On Next:  unknown 
[AIRTHINGS.BLE]-[VERBOSE]-[2018-06-20 08:54:16 +0000] -[MANAGER STATE] On Subscribed 
```

So even if the `CBCentralManagerDelegateWrapper.centralManagerDidUpdateState` returns "poweredOn" its not emitted on the observable since it happens before subscription.
But its still weird, because I am using `.startWith(manager.state)`.

I managed to solve this by changing the `CBCentralManagerDelegateWrapper.didUpdateState` from a `PublishSubject` to a `ReplaySubject` with buffer size of 1. But I don't know if this is correct.

paweljaneczek, 2018-06-20:
@aleksanderaleksic it looks like problem here may be with using different queue's for `CentralManager` (`DispatchQueue(label: "airthings.ble")`) and for `scan(for:)`methods.

In that case it is possible that between `.startWith(manager.state)` and subscribe `CentralManager` is receiving new state, that finally won't be catched by that's observable chain.

We cannot do anything about that, it should be fixed in your code.

aleksanderaleksic, 2018-06-20:
Ahaa, that makes sense. So I should rather run the central manager on main and rather use the `observeOn(sheduler)`?

aleksanderaleksic, 2018-06-21:
You where right! It was observing from main thread, but the Central manager where using the `airthings.ble` thread. I ended up wrapping the contents of scan function inside `queue.sync {}`

closed 2018-06-21 by aleksanderaleksic

paweljaneczek, 2018-06-21:
great! I am happy that I could help you!

c0diq, 2018-11-08:
One way I found to solve it also is to start the CentralManager with a suspended queue. Then resume it after the `observeState` subscription is setup. This guarantees that no matter what queue you're observing it on, it is received.

jordenchang55, 2018-11-19:
@aleksanderaleksic 
I tried you solution with `subscribeOn(scheduler)`. But it still starts with unknown.
Sometime it is followed by `poweredOn`, but most of time, it just keeps unknown state.
Do you have any idea?
```swift
let queue = DispatchQueue(label: "manager.queue")
let sheduler = SerialDispatchQueueScheduler(queue: queue, internalSerialQueueName: "manager.sheduler");

self.centralManager = CentralManager(queue: queue, options: nil);
self.managerStateObservable = self.centralManager.observeState()
      .startWith(self.centralManager.state)
      .share(replay: 1, scope: .forever)
      .subscribeOn(sheduler);
```

aleksanderaleksic, 2018-11-20:
Maybe its because the manager is not created in the same queue? 
like this:

```
let queue = DispatchQueue(label: "manager.queue")
let sheduler = SerialDispatchQueueScheduler(queue: queue, internalSerialQueueName: "manager.sheduler");

var centralManager: CentralManager!
queue.sync {
   centralManager = CentralManager(queue: queue, options: nil);
}

self.managerStateObservable = centralManager.observeState()
      .startWith(self.centralManager.state)
      .share(replay: 1, scope: .forever)
      .subscribeOn(sheduler);
```

Its not prettry but just try and se if it helps 😄

jordenchang55, 2019-04-30:
I finally figured out a solution.
It might be not  perfect but worked for me.
```swift
 centralManager = CentralManager(queue: queue, options: nil);
 self.managerStateObservable = Observable.just(0)
      .map({ i in centralManager.state})
      .flatMap { centralManager.observeState()
          .startWith($0)
      }
      .subscribeOn(sheduler);

```