Polidea/RxBluetoothKit #270: Question about write with response semantics

DevAndArtist, 2018-06-21:
Today I was investigating a few issue in our app and had to turn on verbose logs. I noticed on a write that the log contained a strange value.

```swift
func peripheral(
  _ peripheral: CBPeripheral,
  didWriteValueFor characteristic: CBCharacteristic,
  error: Error?
) {
  RxBluetoothKitLog.d("""
    \(peripheral.logDescription) didWriteValueFor(for:\(characteristic.logDescription),
    value: \(String(describing: characteristic.value?.logDescription)),
    error: \(String(describing: error)))
    """)
  peripheralDidWriteValueForCharacteristic.onNext((characteristic, error))
}
```

My questions are:
- Why are you logging `characteristic.value` after the write?
- Is the `value` supposed to be updated after the write?
- If yes, is it by default?
- If it's a default, then would it make sense also to emit a value update notification?

paweljaneczek, 2018-06-22:
We haven't thought about logging of the value before. Probably someone that have added that logging though that it will have updated value.
I cannot say if value supposed to be updated after the write - it is not written in Apple's documentation so we would need to test it to check what behaviour there is.
Have you maybe checked this? Also is that a problem that library is logging that value?

> If it's a default, then would it make sense also to emit a value update notification?

Not sure if I understand what you mean but logging and emitting new value is done in `peripheral(_ peripheral: CBPeripheral, didUpdateValueFor characteristic: CBCharacteristic, error: Error?)`

DevAndArtist, 2018-06-22:
No issues with library, I just was wondering. Some of our characteristics were adopted to follow a reactive like behavior to emit a new notification when they were updated.

While investigating a few things I was confused by the verbose logs since I myself do log the value I will write to the characteristic but the log from your library reported a different value.
It was more like: Will write `03` and then did write `ff000003` (made up numbers). I drilled down the implementation found that you're just writing the current value.

Then I thought maybe I don't know all semantics on how writes work asked you if maybe it's by default that the value of a characteristic is re-read after a write happens.

I think I can now answer that question by myself: A characteristic can be write-only or notifying-write-only which makes a updating of the current value impossible.

Thank you for taking the time for the short discussion. I think you can close the issue now. ;)

paweljaneczek, 2018-06-22:
I felt like it is more like one way discussion, but yeah, thanks for it ;) Good to know that things.
Closing ;)

closed 2018-06-22 by paweljaneczek