Polidea/RxBluetoothKit #272: Memory leaks (potentially a Swift bug)

DevAndArtist, 2018-06-24:
**Describe the bug**
Please read this [thread](https://forums.swift.org/t/unexpected-memory-leak-dependent-on-3rd-party-code/13964).

**To Reproduce**
Everything mentioned in the linked thread.

**Expected behavior**
This should not leak.

**Environment:**
 - Device: iPhone6 Plus
 - OS: iOS 11.3
 - Library version: 5.1.2
 - Swift version: Apple Swift version 4.1.2 (swiftlang-902.0.54 clang-902.0.39.2)

---

I only used `CentralManager` but I've seen that the new `PeripheralManager` has similar API which would potentially leak as well.

**Workaround:**

Instead of nesting closures like:

```swift 
.subscribe(onNext: { handler($0) })
```

We can pass the escaping `handler` closure directly to `subscribe` since the function types should already match up.

```swift
.subscribe(onNext: handler)
```

paweljaneczek, 2018-06-25:
Hey, could you tell what is leaking? CentralManagerRestoredState? CentralManager?

paweljaneczek, 2018-06-25:
Is it possible for you to make a screenshot of memory graph debugger?

DevAndArtist, 2018-06-25:
Not at this moment. I'll setup a small project in the next few days where it's reproducible and do some screenshots of the debugger and save the memory graph. (To me it looks like a bug in Swift, but I also let you guys know.)

paweljaneczek, 2018-06-25:
Great, looking forward for this 👍

DevAndArtist, 2018-06-25:
Small [update](https://forums.swift.org/t/unexpected-memory-leak-dependent-on-3rd-party-code/13964/2), found the same issue in our code base which is unrelated to `RxBluetoothKit`. Seems like it has an issue with nested functions.

paweljaneczek, 2018-07-23:
@DevAndArtist do you have any update on this?

bartoszstelmaszuk, 2018-09-25:
Thanks for discussion!

I will close this issue since there is no update and it looks like not related directly with RxBluetoothKit library.

closed 2018-09-25 by bartoszstelmaszuk