Polidea/RxBluetoothKit #273: Move Fake classes to new RxBluetoothKitTest framework

c0diq, 2018-06-26:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
There are a number of classes for testing RxBluetoothKit that would be useful to projects using RxBluetoothKit to test their implementation. For example, all the FakeXX classes (FakeCentralManager, etc..), Utilities, etc.. used by the RxBluetoothKit Tests. 
Similar to how Rx has a RxTest framework, it would be great if there was a RxBluetoothKitTest framework that could be used externally to RxBlueoothKit.

**Describe alternatives you've considered**
It is possible to directly link with RxBluetoothKit source code when building RxBluetoothKit from sources but that wouldn't work if binaries are directly linked since these classes are not part of the RxBluetoothKit framework.

paweljaneczek, 2018-06-27:
For now, RxBluetoothKit tests are done in a away that they are not using `RxBluetoothKit` classes at all, so that classes won't help with app testing.

What we are doing is, on each build, we are creating `RxBluettothKit` files copies and changing `CoreBluetooth` references into our generated mocks (mocks are generated with use of swinject).
`CoreBluetooth` test copies are named with `_` prefix, e.g. `_CentralManager`, `_BluetoothState`.
Those copies are named with `_` prefix and are added to `RxBluetoothKit` Test target. 
`CoreBluetooth` mocks are named with `Mock` suffix, e.g. `CBCentralManagerMock`, `CBPeripheralMock`.
It gives us possibility to write tests for our library easily, without adding any complicated abstraction in library.

With that configuration I don't think that publishing that classes as separate library would make any sense. It won't help user with mocking or testing.

What we could do is to make all `RXBluetoothKit` classes implements their own protocols (e.g. `CentralManager` will implement `CentralManagerType`, so that developer who would use our library could create mocks with use of those protocols, and in a result he will be able to write test in an easy way. 

What you think? Or maybe you were thinking about different approach?

c0diq, 2018-08-06:
Any progress on changing RxBluetoothKit to implement own protocols?

closed 2019-05-16 by c0diq