Polidea/RxBluetoothKit #274: peripheral:didDiscoverServices not implemented

aleksanderaleksic, 2018-06-27:
**Describe the bug**
I have a Peripheral that I have been connected to and now try to connect again after a disconnect.
But when I reconnect and try to discoverServices, I get this error from CoreBluetooth:
`[CoreBluetooth] API MISUSE: Discovering services for peripheral <CBPeripheral: 0x1c811c320, identifier = F33ABC8C-4A72-5FCB-C6D0-91708AB10A3B, name = Airthings Node, state = connected> while delegate is either nil or does not implement peripheral:didDiscoverServices:`

**To Reproduce**
Steps to reproduce the behavior:
1. Scan
2. Connect
3. Discover Services
4. Discover Characteristics
5. Disconnect
6. Scan
7. Connect
8. Discover Services

**Expected behavior**
I expect it to scan and return the services.

**Environment:**
 - Device: iPhone8
 - OS: iOS 11.2.6
 - Library version: 5.1.2
 - Swift version 3.2

aleksanderaleksic, 2018-06-27:
Mhm, I am actually using another Library that I provide the CBPeripheral to. 
But shouldn't the delegate get set again when I connect to it?

aleksanderaleksic, 2018-06-27:
I think i see ut now. If i call the attach method om the peripheral object, that will "reset" the peripheral? Right? 

Cannot test it now, i where just looking through your source code.

aleksanderaleksic, 2018-06-28:
It solved it!

closed 2018-06-28 by aleksanderaleksic

dariuszseweryn, 2018-06-28:
Hello @aleksanderaleksic 

Could you share how did you solve the problem?

paweljaneczek, 2018-06-28:
@aleksanderaleksic yes, `attach()` method is for re attaching RxBluetoothKit delegates to `CBPeripheral`. Great that you solved it :)

aleksanderaleksic, 2018-06-28:
@dariuszseweryn I solved it by calling `peripheral.attach()` on the peripheral object.

Like this (but this is a very simple example):
```
manager.scanForPeripherals(withServices: [SOME UUIDS])
                .flatMap{ $0.peripheral.establishConnection() }
                .flatMap{ $0.attach() }
```
So i do this everytime i connect to a peripheral, just to be shure its set.