Polidea/RxBluetoothKit #278: Add disconnect from peripheral method

larryonoff, 2018-08-10:
`disconnect` method was removed from version 5.0. So developers only can disconnect from peripheral by disposing disposable of `establishConnection` method.

As result developers cannot `disconnect` when connection to peripheral restored via state restoration.

Any thoughts about it?

paweljaneczek, 2018-08-10:
Hey @larryonoff.
In order to disconnect from `Peripheral` that was retrieved via restored state you just need to take `Peripheral` from `CentralManagerRestoredState.peripherals` and do
```swift
let connectedDisposable = peripheral.establishConnection().subscribe(...)
connectedDisposable.dispose()
```

Simply `RxBluettothKit` allows to `establishConnection` for peripherals that were not connected within instance of `CentralManager` (under the hood it checks if device is connected from outside library, and if it is connected that it is returning disconnect observable, so you can disconnect from device via `dispose`)

larryonoff, 2018-08-10:
@pawelByszewski huh. Thank you very much!

I think this should be added into documentations / samples somewhere.

closed 2018-08-10 by larryonoff

paweljaneczek, 2018-08-10:
I agree, will add this soon. Thanks!

larryonoff, 2018-08-10:
@paweljaneczek  One more question in this thread. How to avoid `.peripheralIsAlreadyObservingConnection` error?

paweljaneczek, 2018-08-10:
If you get `.peripheralIsAlreadyObservingConnection` it means that you already have ongoing `establishConnection` observable for that `Peripheral`. 
As I understand you are trying to connect to device that was already connected from different place in code. You should avoid that situations because that can lead to numerous of bugs in your application. 
Best practice is to create some facade or wrapper for you BLE communications and store there disposables for connected peripherals.

larryonoff, 2018-08-10:
Thank you for the feedback. I've already created wrapper for storing connections. I have some ideas how to avoid this error.