Polidea/RxBluetoothKit #296: Observe reconnecting state

gauuni, 2018-10-26:
I have connected bluetooth device then simulated a case that bluetooth device is far away or missing. I could catch event that disconnected. When the device is in range and reconnected I didn't receive anything. Though I have used observeConnect() function.

Please help me out.

    private func obserConnect(for peripheral: Peripheral){
        centralManager.observeConnect(for: peripheral).subscribe(onNext: { [unowned self] (peripheral) in
            self.connectionSubject.onNext(Result.success(peripheral))
            }, onError: { [unowned self] error in
                self.connectionSubject.onNext(Result.error(error))
        }).disposed(by: disposeBag)
        centralManager.observeConnect().subscribe(onNext: { (per) in
            print("centralManager.observeConnect() " + (per.name ?? ""))
        }, onError: { (err) in
            print("centralManager.observeConnect() " + err.localizedDescription)
        }).disposed(by: disposeBag)
    }

      func discoverServices(for peripheral: Peripheral) {
        let isConnected = peripheral.isConnected
        
        let connectedObservableCreator = { return peripheral.discoverServices(nil).asObservable() }
        let connectObservableCreator = {
            return peripheral.establishConnection()
                .do(onNext: { [weak self] _ in
                    self?.observeDisconnect(for: peripheral)
                    self?.obserConnect(for: peripheral)
                })
                .flatMap { $0.discoverServices(nil) }
        }
        let observable = isConnected ? connectedObservableCreator(): connectObservableCreator()
        let disposable = observable.subscribe(onNext: { [weak self] services in
                    self?.discoveredServicesSubject.onNext(Result.success(services))
                }, onError: { [weak self] error in
                    self?.discoveredServicesSubject.onNext(Result.error(error))
                })

        if isConnected {
            disposeBag.insert(disposable)
        } else {
            peripheralConnections[peripheral] = disposable
        }
    }

MishkaBalu, 2020-05-28:
I have the same issue

andrewehab, 2022-01-08:
i have same issue also ?!!
any updated ?!!