Polidea/RxBluetoothKit #297: Discovered Peripheral name is Nil

PaulEhrhardt, 2018-11-07:
**Describe the bug**

I have a Bluetooth device that advertises it's name.
When running a scan/discovery with plain old CoreBluetooth i receive a `CBPeripheral` with the correct name in its `name` property. 

When using RxBluetoothKit for scan/discovery on the same device i receive a `nil` value for the name property.

**To Reproduce**
Steps to reproduce the behaviour:
1. Scan for a device that advertises its name
2. Receive `RxBluetoothKit.ScannedPeripheral`, indicating the successful discovery
3. The underlying `RxBluetoothKit.Peripheral` does not contain a value in the `name` property. Also the wrapped `RxBluetoothKit.Peripheral.CBPeripheral` in this object does not contain a value in its own `name` property. 

**Expected behaviour**
The name property should contain the name of the device. 

#### Please note the different behaviour using the CBManagerDelegate:
When performing a discovery via Apple's CoreBluetooth the `CBPeripheral` in the `centralManager(_:didDiscover:advertisementData:rssi:)` delegate callback has a valid value in it's `name` property

**Environment:**
 - Device: [iPhone 6 - iPhone X]
 - OS: [iOS 12]
 - Library version [5.1.4]
 - Swift version [4.0 - 4.2]

PaulEhrhardt, 2018-11-07:
***Further information***
If add a delay of around 100ms before accessing the `name` property i am finally able to read the name.

dariuszseweryn, 2019-01-11:
This is caused by the iOS vanilla API that may populate name at a later time depending on how it was obtained. The library does not know if a particular peripheral will have a name so I do not expect this issue to be fixable. There may be an additional API for name updates though I suppose. We will get back to this issue next week probably