Polidea/RxBluetoothKit #299: Read or Write to a Characteristic leads to disconnect.

PaulEhrhardt, 2018-11-16:
**Describe the bug**
Using RxBluetoothkit, `read` or `write` to a Characteristic leads to immediate disconnect.
Read/Write action is possible on the same device when directly using Apple CoreBluetooth.

**To Reproduce**

1. Using sample code:
I use the example sequence on to of the [Readme](https://github.com/Polidea/RxBluetoothKit/blob/master/README.md)
```
manager.scanForPeripherals(withServices: [serviceId])
    .take(1)
    .flatMap { $0.peripheral.establishConnection() }
    .flatMap { $0.discoverServices([serviceId]) }
    .flatMap { Observable.from($0) }
    .flatMap { $0.discoverCharacteristics([characteristicId]) }
    .flatMap { Observable.from($0) }
    .flatMap { $0.readValue() }
    .subscribe(onNext: { print("Value: \($0.value)") })
```

2. Using Example App:
I encounter the same disconnect behaviour on read/write value.
[image]

**Expected behavior**
`readValue(...)` and `writeValue(...)` on Characteristic should be possible.

**Environment:**
 - Device: [iPhone6Plus]
 - OS: [iOS12, macOS 10.14.1]
 - Library version [5.1.4]
 - Swift: [4.2]

PaulEhrhardt, 2018-11-19:
Further investigation tuned out that the problem was unfortunately device related. 
I will close this issue as no action is needed from your side.

closed 2018-11-19 by PaulEhrhardt

krutitrivedi, 2019-02-25:
from where you get serviceID because i get error