Polidea/RxBluetoothKit #302: Example App is leaking memory

marcmo, 2018-11-27:
**Describe the bug**
While implementing an app that scans for peripherals, connects, reads, disconnects I found that `RxSwift.Resources.total` increased over time.
Testing with the supplied sample app shows a similar behavior, so I suspect somewhere memory is leaking.

**To Reproduce**
Steps to reproduce the behavior:
1. recompile example app as debug build
(add this to Podfile)
```ruby
if target.name == 'RxSwift'
  target.build_configurations.each do |config|
    if config.name == 'Debug'
      config.build_settings['OTHER_SWIFT_FLAGS'] ||= ['-D', 'TRACE_RESOURCES']
    end
  end
end
```
2. narrow down `startScanning` to only find 1 peripheral, e.g.
`scanForPeripherals(withServices: [CBUUID(string: "c446")])`
3. add debug output to `scanAction`
```swift
func scanAction() {
    if isScanning {
        bluetoothService.stopScanning()
        print("scanning stopped...[\(RxSwift.Resources.total)]")
    } else {
        bluetoothService.startScanning()
        print("scanning started...[\(RxSwift.Resources.total)]")
    }
    isScanning = !isScanning
}
```
4. add item to dispose bag in `PeripheralServicesViewModel`
```swift
disposeBag.insert(Disposables.create {
    print("disposeBag fired...[\(RxSwift.Resources.total)]")   
})
```
5. launch app and start scanning, connect, read characteristic, disconnect, stop scan, scan, stop scan, ...
```
scanning started...[136]
disposeBag fired...[206]
scanning stopped...[184]
scanning started...[242]
scanning stopped...[187]
scanning started...[245]
2018-11-27 08:28:39.907521+0100 ExampleApp[10168:651609] Warning: Attempt to present <UIAlertController: 0x109013800>  on <ExampleApp.PeripheralServicesViewController: 0x10c8203d0> which is already presenting (null)
disposeBag fired...[239]
scanning stopped...[217]
scanning started...[275]
scanning stopped...[220]
scanning started...[278]
scanning stopped...[223]
scanning started...[281]
scanning stopped...[226]
scanning started...[284]
scanning stopped...[229]
```

**Expected behavior**
the `RxSwift.Resources.total` count should always go back to the same value

**Environment:**
 - Device: [iPhone 8]
 - OS: [iOS12.1]
 - Library version [5.1.4]
 - Swift version [4.2]

marcmo, 2019-01-09:
ping...are you still working on this library / maintaining it? would appreciate a comment from your side since this is not just a "my app is not working" issue but actually a sever problem IMHO?

dariuszseweryn, 2019-01-11:
Hello @marcmo 
Sorry for such a delay — Yes, we are still maintaining this library. We will look into this issue somewhere next week
Best Regards

marcmo, 2019-01-31:
Hi @dariuszseweryn, where you able to check my example?

marcmo, 2019-03-28:
hello @dariuszseweryn , could you please at least tell me if you can reproduce this bug? it has been a long time since I reported it...

minixT, 2019-03-29:
hi @marcmo, I can reproduce this on my side. I'm working on finding the reason of this behaviour.

nethergrim, 2019-10-07:
Hi folks, any progress on that? We're facing the same issue with react-native-ble-plx