Polidea/RxBluetoothKit #303: Working with group of BLE devices

rameswar54, 2018-12-12:
**Describe the bug**
I'm interested to use this library for bluetooth work in my app, however, pardon me for the lack of experience with Rx if my question is naive.

This is not a bug but just a question for usage of the library and I've checked out the complete documentation. Most of the places the examples are shown to just to scan for peripherals and take(1) and then work with that particular peripheral.

Mine is a bit of different scenario. I have to discover and connect to almost all of the peripherals and perform batch operation like updating a particular characteristic with certain preference in all of the peripherals.

Would be glad if anyone can show a minor snippet or just point in the right direction how to achieve it with this library. Thanks.

rameswar54, 2018-12-17:
Can anyone help on this?

dariuszseweryn, 2019-01-11:
Pardon with this pseudocode but I am not too familiar with the exact API names. But it should be something like this:
```
scanDevices() // start scanning
    .filter(onlyNeededDevices) // get only needed peripherals
    .map(scanResultToPeripheral) // map scan results to peripheral
    .distinct() // emit a particular peripheral only once
    .flatMap(peripheral -> // doe every peripheral
        peripheral.establishConnection() // start a connection
            .flatMap(peripheral -> peripheral.writeCharacteristic()) // when connection is ready write what you want
            .take(1) // after the first write unsubscribe from the chain started by  peripheral.establishConnection() which means disconnect from the peripheral
    )
    .subscribe() // start the flow
```

Does it help?

closed 2019-02-02 by rameswar54